UniversalRobots/Universal_Robots_ROS_Driver #568: Proposal: allow empty `stopped_controllers` arg

adamheins, 2022-08-24:
In my setup I generally know what controller I want to use, and don't want to load extra ones that I don't. Thus I want to be able to not load any stopped controllers, but this is currently not allowed: the line
```
  <node name="ros_control_stopped_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="--stopped $(arg stopped_controllers)" />
```
in `launch/ur_control.launch` will complain when `stopped_controllers` is empty.

An easy fix would be to change the above to something like
```
  <node name="ros_control_stopped_spawner" pkg="controller_manager" type="spawner" respawn="false"
    output="screen" args="--stopped $(arg stopped_controllers)" unless="$(eval arg('stopped_controllers') == '')"/>
```
which I would be happy to do if people think it is useful.

fmauch, 2022-09-19:
I absolutely don't see any harm in doing so, although I also do not see many benefits from not loading them. Probably you want to de-clutter the output of the controller manager?

Anyway, feel free to create a PR with that change, as you already suggested.

adamheins, 2022-09-22:
Resolved by #572

closed 2022-09-22 by adamheins