UniversalRobots/Universal_Robots_ROS_Driver #569: could not get fresh data package from robot

mathgameapi, 2022-09-07:
# Summary

Your issue may already be reported!
Please search on the [issue track](../) before creating one.

The Ur5e is connected to switch with wired cable. The PC is connected to switch with wired cable. But i got the error "could not get fresh data package from robot" at random time.

# Versions

- ROS Driver version: v2.0.0
- Affected Robot Software Version(s): URSoftware 5.11.5.1010327 (Oct 15 2021)
- Affected Robot Hardware Version(s):
- Robot Serial Number: 20215502112
- UR+ product(s) installed: UR5e
- URCaps Software version(s): 1.0.5

fmauch, 2023-01-02:
Hi there, thank you for reporting this issue. Without further information this looks like an issue related to real-time performance. We suggest using a separate PC with a dedicated network interface for the robot connection running a realtime kernel.

From our experience you can get along using a non-realtime kernel for development environments, but you'll have to expect regular disconnects since the robot controller expects to get a command from the ROS side in each control cycle (every 2ms in case of an e-Series robot).

Since you don't get new data from the robot I would assume that either your network is overloaded with data (in case you're running over a shared network interface / switch) or it is indeed a cpu load problem on your ROS pc. 

If this is a development system and the driver recovers from occasional "could not get fresh data package from robot" errors, you can decide whether you accept to live with that or whether it would be worth it changing your setup.

evacheung0929, 2023-03-28:
Hi, I am using ROS Melodic on VM to control two real robots and I am facing the same issue here. Is there any work around by any chance? Or what is the recommended PC/ Desktop to avoid that problem in the future please? Thank you

closed 2023-07-27 by github-actions[bot]