UniversalRobots/Universal_Robots_ROS_Driver #571: UR5e won't move with published topic "/joint_group_vel_controller/command"

iiceberrg, 2022-09-17:
# Summary

UR5e won't move with published topic "/joint_group_vel_controller/command"

I have referred back to #231 and used the code from https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/231#issuecomment-688415091 to see if it works, but the robot couldn't move, too.

I tried to publish an array of velocities to the topic "/joint_group_vel_controller/command" to control and move all six joints of the robot arm. The robot stays stationary even rostopic echo /joint_group_vel_controller/command shows that the values are being published.

[image]

# Versions
 - ROS Driver version: melodic
 - Affected Robot Software Version(s): URSoftware 5.5.0.82143
 - Affected Robot Hardware Version(s): UR5e

# Impact

UR5e real robot is not able to move at all with the topic /joint_group_vel_controller/command but is able to move with /scaled_pos_point_traj_controller/command

# Issue details

I have a Python script showed below that publishes an array of velocities to the topic /joint_group_vel_controller/command. The code is able to move the robot in Gazebo simulation but not with the real hardware. 

`#!/usr/bin/env python`

`import rospy`
`from std_msgs.msg import Float64MultiArray`
`import numpy as np`

`class MoveUR5e():`
    
    # Spin shoulder_pan_joint constantly

    def __init__(self, angular_vels):

        self.angular_vels = angular_vels

        self.pub = rospy.Publisher('/joint_group_vel_controller/command', Float64MultiArray, queue_size=10)

    def spin_base_joint(self):
        angular_vels = self.angular_vels
        self.msg = Float64MultiArray()
        self.msg.data = angular_vels
        # self.msg.layout.data_offset = 0
        self.pub.publish(self.msg)

`if __name__ == "__main__":`

    try:
        rospy.init_node('test_node', anonymous=True)
        move_ur = MoveUR5e(angular_vels = np.array([-0.2, 0.0, 0.0, 0.0, 0.0, 0.0]))
        rate = rospy.Rate(50)

        while not rospy.is_shutdown():
            move_ur.spin_base_joint()
            rate.sleep()
            
    except rospy.ROSInterruptException:
        pass `

Below is what shown in the terminal when I roslaunch the gazebo simulation:
[image]

Below is what shown in the terminal when I roslaunch ur_robot_driver:
[image]

Below is the rostopic list of the ur_robot_driver:
[image]

## Project status at point of discovered
*When did you first observe the issue?*
 - *Few months ago*
 - *I tried many times and it still doesn't work*

## Steps to Reproduce
- UR5e robot connected to NETGEAR GS305P and to the computer.
- On the Teach Pendant, External Control - Custom port: 50002
- Created a launchfile for the code mentioned above.
- Clone and roslaunch ur_robot_driver for ur5e_bringup as stated in the repository.
- Launch the code and cry because the robot won't move

## Expected Behavior
The robot arm should move accordingly to the joint velocities assigned in the Python script. It is able to move in Gazeboso it should also work for ur_robot_driver with the same topic.

## Actual Behavior
The robot won't move at all, no response. But the rostopic echo /joint_group_vel_controller/command is showing that it is publishing data.

Below is the rostopic echo /joint_group_vel_controller/command
[image]

Below is the rqt_graph in case you suspect I wired it wrongly:
[image]

shukla-shubh, 2022-09-19:
Hi @iiceberrg,

I am also attempting the same problem, but first in simulation. Can you share the config/*_controllers.yaml file which defines the joint_group_vel_controller?

iiceberrg, 2022-09-21:
joint_group_vel_controller:
  type: velocity_controllers/JointGroupVelocityController
  joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint
  gains:
    #!!These values have not been optimized!! (taken from ur_robot_driver)
    shoulder_pan_joint: {p: 5000.0,  i: 0.5, d: 10.0, i_clamp: 1}
    shoulder_lift_joint: {p: 5000.0,  i: 0.5, d: 10.0, i_clamp: 1}
    elbow_joint: {p: 5000.0,  i: 0.5, d: 10.0, i_clamp: 1}
    wrist_1_joint: {p: 5000.0,  i: 0.5, d: 10.0, i_clamp: 1}
    wrist_2_joint: {p: 5000.0,  i: 0.5, d: 10.0, i_clamp: 1}
    wrist_3_joint: {p: 5000.0,  i: 0.5, d: 10.0, i_clamp: 1}

closed 2022-09-21 by iiceberrg

fmauch, 2022-09-21:
@iiceberrg did you get it running? I assume that a controller switch was missing to get it to work?

shukla-shubh, 2022-09-21:
@iiceberrg, thanks for sharing. I found the controller details in the ur_robot_driver. I was using the ur_gazebo package earlier.

My objective is to make the wrist3 run at constant velocity, however due to the joint constraints, it is making only 2 complete rotations. Is there any way to make the wrist3 rotate continuously at any input velocity?

I saw something called screwdriver operation, but couldn't find much information on how to run that.

@fmauch, can you help with this?

fmauch, 2022-09-21:
No, only the ur3(e) can turn indefinitely. It is a hardware restriction, as the ur3 is built differently.