UniversalRobots/Universal_Robots_ROS_Driver #579: Teach pendent freezes when connecting to ROS

SethCHPetersen, 2022-10-11:
Every time I try to connect my UR3e to the ROS driver using 

roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=128.105.49.126

and also using the external control URCap, the teach pendant freezes.

I have updated the Tp to the latest versions and also updated all of the Ros drivers. I have seen this issue posted here already but it was from a couple of months ago and did not have a solution. Has anyone found a solution to this issue?

fmauch, 2022-10-11:
For reference, the other issues mentioning freezes are #508 and #360.

Just to make sure: Once you start `roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=128.105.49.126` the TP freezes? (128.105.49.126 being the IP address of your robot).

As we establish a couple of connections, we can maybe debug this a bit further. The driver connects to the following interfaces:

- primary interface on port 30001
- secondary interface on port 30002
- RTDE interface on port 30004
- Dashboard interface on port 29999

The dashboard interface can easily be tested standalone by running

```
rosrun ur_robot_driver dashboard_client _robot_ip:=128.105.49.126
```

If this runs fine, we can check the others, which will be a bit more complicated. You'll need to make an all-src build as explained in the README containing the ur_client_library. Before building, replace the hardcoded IP addresses in https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/boost/examples/full_driver.cpp, https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/boost/examples/primary_pipeline.cpp, https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/boost/examples/rtde_client.cpp.

Then, build your workspace using `catkin_make_isolated`. Once built and you sourced your workspace, navigate the the ur_client_library's source folder (The repository you cloned). In there, run the following three commands, checking for freeze after each of them. You'll have to stop them by pressing CTRL+C manually. 

```
../../build_isolated/ur_client_library/devel/examples/primary_pipeline_example # this apparently has no output
../../build_isolated/ur_client_library/devel/examples/rtde_client_example
../../build_isolated/ur_client_library/devel/examples/driver_example
```

Any of them should probably cause the freezing, if this is a systematic error.

SethCHPetersen, 2022-10-11:
Thanks for getting back to me so quickly. That is correct that the TP freezes, it brings up the "starting program" dialog box and then freezes. I am unable to un-freeze it by ctl-C the ros launch. I attempted to build the all source build however when using catkin_make_isolated I get an error saying 
"failed to process package 'ur_robot_driver'"
despite the fact that I can build that package just fine when following the non-source instructions. So I deleted that  package from that directory and then tried running the executables you suggested with then running the UR robot driver launch file from the non-souce directory. I was able to run the first executable and it connected but wasnt outputting anything and when I tried to connect the robot it again froze. As for the other two executables they wouldnt run at all saying 

./driver_example 
./driver_example: symbol lookup error: ./driver_example: undefined symbol: _ZN4urcl8UrDriverC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES8_S8_S8_St8functionIFvbEEbSt10unique_ptrINS_13ToolCommSetupESt14default_deleteISD_EES8_jjidbS8_jj

 ./rtde_client_example 
./rtde_client_example: symbol lookup error: ./rtde_client_example: undefined symbol: _ZN4urcl14rtde_interface10RTDEClientC1ENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS_4comm9INotifierERKS7_SC_d

SethCHPetersen, 2022-10-11:
The dashboard interface seemed to work fine aswell.

SethCHPetersen, 2022-10-11:
Also just foud that if I try to save a program with an external control URcap command in it, the TP endlessly says "in progress... please wait" and never actually saves. In order to get it to stop I have to shut off the robot and reboot.

fmauch, 2022-10-12:
> Thanks for getting back to me so quickly. That is correct that the TP freezes, it brings up the "starting program" dialog box and then freezes. I am unable to un-freeze it by ctl-C the ros launch.

That does not seem right. Only starting the driver should not make any program start on the robot. That's why I asked. You are not starting any program on the robot controller and you are not using headless mode?

> Also just foud that if I try to save a program with an external control URcap command in it, the TP endlessly says "in progress... please wait" and never actually saves.

That's rather interesting. While saving, the robot tries to connect to the driver to query the URScript code. However, that should either succeed or fail. Maybe you should see whether there is a firewall active on your ROS machine. I've seen firewalls cause weird behavior in multiple ways.

The symbol lookups you mentioned could come from having the binary package installed on your system, as well. You can try to uninstall `ros-noetic-ur-client-library` to mitigate this. Of course, reinstall it, when you want to use your non-source workspace again.

closed 2023-02-22 by github-actions[bot]