UniversalRobots/Universal_Robots_ROS_Driver #581: Multi robot control broken, Failed to bind socket for port 50004 to address

LorenzG1, 2022-10-18:
When launching a configuration with multiple robots a port conflict on 50004 is reported. The exact message is:
  what():  Failed to bind socket for port 50004 to address. Reason: Address already in use: Address already in use

Reason for this is the addition of a script_command_port. (https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/111)

To make it work the script_command_port has to be added to ur_control.launch and hardware_interface.cpp. Code already in place so I can make a PR for this.

LorenzG1, 2022-10-18:
No problem. Changes are...
ur_robot_driver/launch/ur_control.launch, added two lines:
16   `<arg name="script_command_port" default="50004" doc="PPort used for forwarding script commands to the robot. The script commands will be executed locally on the robot."/>`
43 `    <param name="script_command_port" value="$(arg script_command_port)"/>`

ur_robot_driver/src/hardware_interface.cpp, added following lines:
112 `  // Port that will be opened to send trajectory points from the driver to the robot
  int script_command_port = robot_hw_nh.param("script_command_port", 50004);`
Added the script_command_port parameter to the call of ur_driver_.reset at line 283:
`    ur_driver_.reset(new urcl::UrDriver(
        robot_ip_, script_filename, output_recipe_filename, input_recipe_filename,
        std::bind(&HardwareInterface::handleRobotProgramState, this, std::placeholders::_1), headless_mode,
        std::move(tool_comm_setup), (uint32_t)reverse_port, (uint32_t)script_sender_port, servoj_gain,
        servoj_lookahead_time, non_blocking_read_, reverse_ip, trajectory_port, script_command_port));`

gavanderhoorn, 2022-10-18:
@miguelprada opened #582 a couple of hours ago. That shows the changes necessary I believe.

#567 also contains the necessary change (confusingly, as it isn't mentioned at all).

fmauch, 2022-10-18:
Yes, this was recently introduced. We explicitly took care about backwards compatibility, but we didn't think of multi-robot setups where this would obviously cause issues. Sorry for the inconvenience caused. As soon as the library is also synced for melodic, we will merge #567 (with some additional things from #582).

aatb-ch, 2022-11-05:
just hit this issue after upgrading as well, for anyone looking for a quick workaround before the patch is merged, simply pick higher port values for conflicting robot to have it back up right away (we used defaults 50001, 50002, 50003 for robot A and 50004, 50005, 50006 for robot B, simply moved up to 50005, 50006, 50007, dont forget to update the port number in External Control URcap).

aatb-ch, 2022-11-05:
My bad.. silly me just realized the issue is not only to have a secondary robot not using 50004 port but to open twice the currently hardcoded `script_command_port` on each robot, so indeed the above fix *must* be implemented to be able to run the driver on multiple robots *on the same machine*.

closed 2022-11-30 by fmauch

catmulti7, 2023-02-08:
> Hi everyone, after solving the issue regarding script_command_port, I ran into another one recently. The message just appeared as follows:
> 
> what(): Failed to bind socket for port 50004 to address. Reason: Address already in use: Address already in use
> 
> how to solve this issue? Did it happen to anyone else? thanks in advance

Hi @KhalilAbuibaid , I ran into the same issue with ROS Noetic, everything is up-to-date (driver and client_library), unfortunately I have no idea about what caused this issue

KhalilAbuibaid, 2023-02-09:
@catmulti7 after the update, it should work fine. However, my problem was that I hav forgot to roslaunch the robot controller with the right script_command IP, which has led to be chosen by default 50004.