UniversalRobots/Universal_Robots_ROS_Driver #586: TF_REPEATED_DATA ignoring data with redundant timestamp for frame tool0_controller at time xxxx according to authority ur_hardware_interface

cambel, 2022-11-17:
# Summary

We have a dual arm set up with two UR5e robots. Since migrating our system to ROS Noetic we experience a spam of warnings coming from `ur_hardware_interface`.

# Versions
 - ROS Driver version: noetic (1.15.14)
 - Affected Robot Software Version(s): 5.10.0.106288
 - Affected Robot Hardware Version(s):
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): 1.0.5

# Impact
We have a compliance controller streaming motion commands. The performance significantly decreases when the spam of warnings are printed.

## Steps to Reproduce
Here is our launch file:
https://github.com/o2ac/o2ac-ur/blob/noetic-devel/catkin_ws/src/o2ac_scene_description/launch/osx_arms_bringup.launch

## Expected Behavior
No spam of warning messages.

## Actual Behavior
The following is a sample of the messages that are spammed in the terminal.
```shell
[ WARN] [1668669563.967702653]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame tool0_controller at time 1668669563.965094 according to authority /a_bot/ur_hardware_interface
[ WARN] [1668669564.233413326]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame tool0_controller at time 1668669564.231068 according to authority /b_bot/ur_hardware_interface
```
In our case the namespace for the robots are `a_bot` and `b_bot`.

The spam of warning messages is not continuous but it occurs very frequently even when there is no command sent to the robots.

gavanderhoorn, 2022-11-17:
This is https://github.com/ros/geometry2/issues/467.

Are the TF frames prefixed?

cambel, 2022-11-18:
Hi @gavanderhoorn 

My bad, for lack of use of the `tool0_controller`, we didn't set up the parameter `tf_prefix` before. Adding a different `tf_prefix` to each robot seems to fix the problem.

closed 2022-11-18 by cambel