UniversalRobots/Universal_Robots_ROS_Driver #589: Error while launching moveit with real UR5e robot and ROS noetics

stevensu1838, 2022-11-23:
# Summary

I followed the documentation to install [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) and I can successfully launch roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.177

And rqt_joint_trajectory_controller gui works well. I can drag the robot with the sliders for each joint. 

However, it always throw errors when I run the following in the second terminal to bring up moveit: roslaunch ur5e_moveit_config ur5e_moveit_planning_executing.launch

`$ roslaunch ur5e_moveit_config ur5e_moveit_planning_executing.launch
RLException: [ur5e_moveit_planning_executing.launch] is neither a launch file in package [ur5e_moveit_config] nor is [ur5e_moveit_config] a launch file name
The traceback for the exception was written to the log file
`
Did I miss anything? Please forgive me if the question is silly. Just don't know what I did wrong. Cheers.

By the way, I've noticed the documentation is for controlling the real robots, however, with this Universal_Robots_ROS_Driver, how do I run a UR5e robot in simulation? 

# Versions
 - ROS Driver version:
 - Affected Robot Software Version(s): URSoftware 5.12.2.1101534
 - Affected Robot Hardware Version(s):
 - Robot Serial Number: 20225501705
 - UR+ product(s) installed:
 - URCaps Software version(s): [v1.0.5](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/releases/tag/v1.0.5)

RobertWilbrandt, 2022-11-24:
You are trying to start a launch file that does not exist, the correct command should be:
```console
roslaunch ur5e_moveit_config moveit_planning_execution.launch
```
MoveIt usage is described in the [documentation](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/usage_example.md#control-the-robot-using-moveit). Does this work for you?

As for simulation, you have two options:
- Use `ursim` as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/usage_example.md#control-the-robot-using-moveit). This is as close to using a real robot as possible.
- Use the `ur_gazebo` package. This is described in the [universal_robot](https://github.com/ros-industrial/universal_robot/blob/melodic-devel/README.md) README.

stevensu1838, 2022-11-24:
@RobertWilbrandt 
It works. Thanks a million. Could you please also take a look at [this issue](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/590#issue-1463202793)?

stevensu1838, 2022-11-26:
@RobertWilbrandt Hi I've followed MoveIt usage is described in the [documentation](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/usage_example.md#control-the-robot-using-moveit). 
Terminal one: roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.177

Terminal two: roslaunch ur_robot_driver example_rviz.launch
Terminal three: rosrun ur_robot_driver test_move

And I get the following errors in terminal one. Can you please help?
[ERROR] [1669460081.171485561]: A controller named 'scaled_pos_joint_traj_controller' was already loaded inside the controller manager
[ERROR] [1669460081.175082649]: Robot control is currently inactive. Starting controllers that claim resources is currently not possible. Not starting controller 'scaled_pos_joint_traj_controller'
[ERROR] [1669460081.175151547]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
[ERROR] [1669460086.140463291]: Can't accept new action goals. Controller is not running.

[image]

stevensu1838, 2022-11-26:
And when I Control the robot using MoveIt!

For controlling the robot using MoveIt! start the following three launch files (Each in a separate terminal window):

roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.56.177

roslaunch ur5e_moveit_config moveit_planning_execution.launch

roslaunch ur5e_moveit_config moveit_rviz.launch rviz_config:=$(rospack find ur5e_moveit_config)/launch/moveit.rviz

I get the following errors in terminal one: 

**[ERROR] [1669460448.187771148]: Can't accept new action goals. Controller is not running.
[ERROR] [1669460455.971469826]: Can't accept new action goals. Controller is not running.**

Help please.

fmauch, 2022-11-30:
This is the [first of our FAQ items](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#i-started-everything-but-i-cannot-control-the-robot)

To sum up this and #590 

* start the three commands you wrote in your last post
* press play on the teach pendant
* use MoveIt!

closed 2023-03-31 by github-actions[bot]