UniversalRobots/Universal_Robots_ROS_Driver #590: Connection refused with UR5e and ROS noetic

stevensu1838, 2022-11-24:
The IP of Ubuntu machine is 192.168.0.180
[image]

Set up the IP of UR5e robot, I first got this IP by click DHCP and then made this same IP 192.168.0.177 as Static 

[image]

[image]
[image]

I can ping the UR5e robot (192.168.0.177) from Ubuntu PC (192.168.0.180) with no problem, however, when I click play, It throws the error below on the teach pendant: 
[image]

And in Rviz no UR5e robot shows up. Could you please help? 

# Versions
 - ROS Driver version: 
 - Affected Robot Software Version(s): URSoftware 5.12.2.1101534
 - Affected Robot Hardware Version(s):
 - Robot Serial Number: 20225501705
 - UR+ product(s) installed:
 - URCaps Software version(s): v1.0.5](https://github.com/UniversalRobots/Universal_Robots_ExternalControl_URCap/releases/tag/v1.0.5)

fmauch, 2022-11-30:
Did you run the ros driver on the ROS machine before pressing play as [written in the README](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#quick-start)?

fmauch, 2023-01-02:
Or maybe a more stupid question: Did you start the ROS driver beforehand?

closed 2023-03-15 by stevensu1838