UniversalRobots/Universal_Robots_ROS_Driver #591: Error while Setting up Ubuntu with a PREEMPT_RT kernel

stevensu1838, 2022-11-26:
mech1936:~/rt_kernel_build/linux-5.15.79$ make -j `getconf _NPROCESSORS_ONLN` deb-pkg
make clean
  CLEAN   arch/x86/tools
sh ./scripts/package/mkdebian
  TAR     linux-upstream.tar.gz
origversion=$(dpkg-parsechangelog -SVersion |sed 's/-[^-]*$//');\
	mv linux-upstream.tar.gz ../linux-upstream_${origversion}.orig.tar.gz
dpkg-buildpackage -r"fakeroot -u" -a$(cat debian/arch)  -i.git -us -uc
dpkg-buildpackage: info: source package linux-upstream
dpkg-buildpackage: info: source version 5.15.79-rt54-1
dpkg-buildpackage: info: source distribution focal
dpkg-buildpackage: info: source changed by ysu66 <ysu66@mech1936>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source -i.git --before-build .
 debian/rules clean
rm -rf debian/*tmp debian/files
make clean
 dpkg-source -i.git -b .
dpkg-source: info: using source format '1.0'
dpkg-source: warning: source directory 'linux-5.15.79' is not <sourcepackage>-<upstreamversion> 'linux-upstream-5.15.79-rt54'
dpkg-source: warning: .orig directory name linux-5.15.79.orig is not <package>-<upstreamversion> (wanted linux-upstream-5.15.79-rt54.orig)
dpkg-source: info: building linux-upstream using existing linux-upstream_5.15.79-rt54.orig.tar.gz
dpkg-source: info: building linux-upstream in linux-upstream_5.15.79-rt54-1.diff.gz
dpkg-source: warning: ignoring deletion of file .scmversion
dpkg-source: warning: the diff modifies the following upstream files: 
 .clang-format
 .cocciconfig
 .get_maintainer.ignore
 .mailmap
 CREDITS
 MAINTAINERS
 README
dpkg-source: info: use the '3.0 (quilt)' format to have separate and documented changes to upstream files, see dpkg-source(1)
dpkg-source: info: building linux-upstream in linux-upstream_5.15.79-rt54-1.dsc
dpkg-source: warning: missing information for output field Standards-Version
 debian/rules binary
make KERNELRELEASE=5.15.79-rt54 ARCH=x86 	KBUILD_BUILD_VERSION=1 -f ./Makefile
  HOSTCC  scripts/basic/fixdep
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_32.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_64.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_x32.h
  SYSTBL  arch/x86/include/generated/asm/syscalls_32.h
  SYSHDR  arch/x86/include/generated/asm/unistd_32_ia32.h
  SYSHDR  arch/x86/include/generated/asm/unistd_64_x32.h
  SYSTBL  arch/x86/include/generated/asm/syscalls_64.h
  SYSTBL  arch/x86/include/generated/asm/syscalls_x32.h
  HYPERCALLS arch/x86/include/generated/asm/xen-hypercalls.h
  DESCEND objtool
  DESCEND bpf/resolve_btfids
  MKDIR     /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids//libbpf
  MKDIR     /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids//libsubcmd
<stdin>:1:10: fatal error: libelf.h: No such file or directory
compilation terminated.
  HOSTCC  /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids/fixdep.o
  HOSTCC  /home/ysu66/rt_kernel_build/linux-5.15.79/tools/objtool/fixdep.o
  GEN     /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids/libbpf/bpf_helper_defs.h
  HOSTCC  arch/x86/tools/relocs_32.o
  HOSTCC  arch/x86/tools/relocs_64.o
  HOSTCC  arch/x86/tools/relocs_common.o
  HOSTLD  /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids/fixdep-in.o
  LINK    /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids/fixdep
  HOSTCC  scripts/bin2c
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/sorttable
  HOSTCC  scripts/asn1_compiler
  HOSTCC  scripts/genksyms/genksyms.o
  HOSTLD  /home/ysu66/rt_kernel_build/linux-5.15.79/tools/objtool/fixdep-in.o
  MKDIR   /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids/libbpf/staticobjs/
  LINK    /home/ysu66/rt_kernel_build/linux-5.15.79/tools/objtool/fixdep
  HOSTCC  scripts/selinux/genheaders/genheaders
[... 90 lines cut ...]
/home/ysu66/rt_kernel_build/linux-5.15.79/tools/objtool/include/objtool/elf.h:10:10: fatal error: gelf.h: No such file or directory
   10 | #include <gelf.h>
      |          ^~~~~~~~
compilation terminated.
make[7]: *** [/home/ysu66/rt_kernel_build/linux-5.15.79/tools/build/Makefile.build:97: /home/ysu66/rt_kernel_build/linux-5.15.79/tools/objtool/arch/x86/decode.o] Error 1
make[6]: *** [/home/ysu66/rt_kernel_build/linux-5.15.79/tools/build/Makefile.build:139: arch/x86] Error 2
make[5]: *** [Makefile:56: /home/ysu66/rt_kernel_build/linux-5.15.79/tools/objtool/objtool-in.o] Error 2
make[5]: *** Waiting for unfinished jobs....
  LD      /home/ysu66/rt_kernel_build/linux-5.15.79/tools/objtool/libsubcmd-in.o
  AR      /home/ysu66/rt_kernel_build/linux-5.15.79/tools/objtool/libsubcmd.a
make[4]: *** [Makefile:69: objtool] Error 2
make[3]: *** [Makefile:1403: tools/objtool] Error 2
make[3]: *** Waiting for unfinished jobs....
  LD      /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids/libsubcmd/libsubcmd-in.o
  HOSTLD  scripts/genksyms/genksyms
  AR      /home/ysu66/rt_kernel_build/linux-5.15.79/tools/bpf/resolve_btfids/libsubcmd/libsubcmd.a
make[4]: *** [Makefile:72: bpf/resolve_btfids] Error 2
make[3]: *** [Makefile:1403: tools/bpf/resolve_btfids] Error 2
make[2]: *** [debian/rules:7: build-arch] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:77: deb-pkg] Error 2
make: *** [Makefile:1608: deb-pkg] Error 2
ysu66@mech1936:~/rt_kernel_build/linux-5.15.79$ 

`

stevensu1838, 2022-11-26:
[image]

stevensu1838, 2022-11-26:
[image]

RobertWilbrandt, 2022-11-30:
We cannot help you if you provide nothing more than a console log. Did you follow the [Real-Time setup guide](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md)?

In any way: Provide us with a complete list of command that lead to this error.

closed 2023-02-17 by github-actions[bot]

stevensu1838, 2023-02-24:
Hi all,
I've provided more details [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/618#issue-1595993795). Could you please help?