UniversalRobots/Universal_Robots_ROS_Driver #595: how to control joints with velocity command

7519060, 2022-12-16:
# Summary
Usage of velocity controllers with python

# Versions
 - ROS Driver version:kinetic
 - Affected Robot Hardware Version(s):UR5e and CB3
 - ubuntu 16.04LTS

# Impact
I can't find a way to send velocity control command to each joint using python.
I want to control real UR5e or CB3 with velocity.

# Issue details
Hi, I'm new to ROS and currently started to write a program to control robot arms.
I followed some tutorials and I made some programs to move robot arms to desired joint values or poses with python moveit_commander class.However, I couldn't find a class or module that can send velocity command to each joint. I want to know if there is a class for velocity control.
I found some issues here using JointGroupVelocityController as a topic, but not clearly understood usage of it.
If there is no such class or module, could anyone tell me usage of hw interface? It's a bit confusing since I'm not familiar with ros_control stuff.
Or should I just give up doing this using python and use cpp instead?

Specifically, I'd like to create a program that calculates the angular velocity of each joint every time an image is sent from realsense d435i as a topic in image_raw topic and sends it to CB3/UR5e as a command.

This is my first post.　
Please let me know if there are any deficiencies. 
Thanks in advance.

RobertWilbrandt, 2023-02-13:
Sorry for the late answer. In case this is still useful to somebody:

The `joint_group_vel_controller` is a good way to implement this. It is a `velocity_controllers/JointGroupVelocityController`, meaning that it takes a simple `std_msgs/Float64MultiArray` as input (on topic `/joint_group_vel_controller/command`), interprets each value of the array as one joint velocity and continuously moves the arm as commanded.

If you want to use this controller, the following steps are necessary:
1. Start the driver
2. Stop the `scaled_pos_joint_traj_controller` controller:
```
rosservice call /controller_manager/switch_controller "start_controllers: []
stop_controllers: ['scaled_pos_joint_traj_controller']
strictness: 2
start_asap: false
timeout: 0.0"
```
3. Start the `joint_group_vel_controller` (you can combine this with the previous step):
```
rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_vel_controller']
stop_controllers: []
strictness: 2
start_asap: false
timeout: 0.0"
```
4. You should now be able to command joint velocities. Make sure you are able to stop the motion, as the robot will continue following the command once it was sent. You can do this using:
```
rostopic pub /joint_group_vel_controller/command std_msgs/Float64MultiArray --once "
data: [0.05, 0.0, 0.0, 0.0, 0.0, 0.0]"
```

Be aware that you need to change the controllers back to be able to control the robot using MoveIt again. If you only need to use this type of velocity control, you can tell the driver to exactly load your required controllers:
```
roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=<your robot ip> controllers:="joint_state_controller speed_scaling_state_controller force_torque_sensor_controller joint_group_vel_controller" stopped_controllers:='scaled_pos_joint_traj_controller'
```

On a final note: Be aware that [ROS Kinetic is EOL](http://wiki.ros.org/Distributions), meaning that we won't be able to offer support for problems that are specific to this distriibution. You should consider moving towards ROS Noetic instead.

stevensu1838, 2023-03-17:
@RobertWilbrandt 
Hi, thanks a lot for your reply. It's been very useful. I also want to use the joint_group_vel_controller. However, I hope not to do the following switch every time I launch the robot. I assume the scaled_pos_joint_traj_controller controller is the default controller. May I please ask how I can make he joint_group_vel_controller the default one? Thanks a lot. Cheers. 

    **Start the driver
    Stop the scaled_pos_joint_traj_controller controller:**

rosservice call /controller_manager/switch_controller "start_controllers: []
stop_controllers: ['scaled_pos_joint_traj_controller']
strictness: 2
start_asap: false
timeout: 0.0"

    **Start the joint_group_vel_controller (you can combine this with the previous step):**

rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_vel_controller']
stop_controllers: []
strictness: 2
start_asap: false
timeout: 0.0"

7519060, 2023-03-21:
Thank you so much for your reply @RobertWilbrandt !
I will try this and check whether it is available or not in my environment.

7519060, 2023-05-12:
It worked.
Thank you very much.

closed 2023-05-12 by 7519060

pucciland95, 2024-06-15:
I am sorry that I need to re-open this issue. 
I followed what @RobertWilbrandt advised. Launching:
```
roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=<your robot ip> controllers:="joint_state_controller speed_scaling_state_controller force_torque_sensor_controller joint_group_vel_controller" stopped_controllers:='scaled_pos_joint_traj_controller'
```
However, when I try to move the robot with: 
```
rostopic pub /joint_group_vel_controller/command std_msgs/Float64MultiArray --once "
data: [0.05, 0.0, 0.0, 0.0, 0.0, 0.0]"
```
the robot does not move and does not throw any error.
I noticed then that the driver loads the joint_group_vel_controller but in a 'stopped' state. 
When I try to do:
```
rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_vel_controller']
stop_controllers: []
strictness: 2
start_asap: false
timeout: 0.0"
```
It throws an error regarding the robot not having such an interface. Looking in the ur_description, it is set as default a PositionJointInterface. 
Setting a VelocityJointInterface, however, does not solve the issue., leading to the same error as before when trying to achieve the switch.

Do you have any idea why?

Niccolo

fmauch, 2024-06-19:
@pucciland95 Simply executing the commands you wrote, the controller indeed ends up in the stopped state, since the robot isn't started. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver?tab=readme-ov-file#i-started-everything-but-i-cannot-control-the-robot.

when running 
```
rosservice call /controller_manager/switch_controller "start_controllers: ['joint_group_vel_controller']
stop_controllers: []
strictness: 2
start_asap: false
timeout: 0.0"
```

the driver's output also clearly states

```
[ERROR] [1718786601.391811457]: Robot control is currently inactive. Starting controllers that claim resources is currently not possible. Not starting controller 'joint_group_vel_controller'
[ERROR] [1718786601.391861826]: Could not switch controllers. The hardware interface combination for the requested controllers is unfeasible.
```