UniversalRobots/Universal_Robots_ROS_Driver #599: Error 'Could not correctly initialize robot. Exiting'

7519060, 2022-12-20:
# Summary
I cannot connect UR3 to my PC with ur_robot_driver. 
When I run 'roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.56.1 kinematics_config:=$(rospack find ur_robot_driver)/config/ur3_calibration.yaml ', I got 'Variable 'tcp_offset' not recognized by the robot. Probably your output recipe contains errors' and 'Could not correctly initialize robot. Exiting'.

# Versions
 - ROS Driver version:ubuntu 18.04 + melodic
 - Affected Robot Software Version(s):3.8.0.61336
 - Affected Robot Hardware Version(s):UR3
 - Robot Serial Number:2016331423
 - URCaps Software version(s):externalcontrol-1.0.5.urcap 

# Impact
I cannot use ur_robot_driver becaues of the error. Error messages says 
`
[FATAL] [1671551631.516733597]: Variable 'tcp_offset' not recognized by the robot. Probably your output recipe contains errors
[ERROR] [1671551631.516866722]: Could not correctly initialize robot. Exiting
`
I'm new to ROS. If something is wrong with this post, please tell me.
I would appreciate it if you could give me a solution.

# Issue details
In order to use velocity_controllers/JointGroupVelocityController type as a topic, I needed to upgrade my driver for universal robot ur3. I followed steps written in README except realtime kernel setups(I couldn't locate the right key using 'gpg2 ' ). 
From terminal, when I run the command 'roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.56.1 kinematics_config:=$(rospack find ur_robot_driver)/config/ur3_calibration.yaml ', I got 
`
sgn@ren:~/vs_ws$ roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.56.1 kinematics_config:=$(rospack find ur_robot_driver)/config/ur3_calibration.yaml [script_sender_port:=SCRIPT_SENDER_PORT]
... logging to /home/sgn/.ros/log/7b359b70-807e-11ed-87f9-f02f74a14b89/roslaunch-ren-7989.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ren:42793/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 125
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
 * /joint_based_cartesian_traj_controller/type: position_controll...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
[... 309 lines cut ...]
default gateway:192.168.56.11

Ecternal Control network settings:
Host IP:192.168.56.1
Custom Port:50002

## Project status at point of discovered
*When did you first observe the issue?*
 - yesterday
 - I worked hard to find similar error but couldn't find one.

## Steps to Reproduce
I also tried the command 
'
roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.56.1 kinematics_config:=$(rospack find ur_robot_driver)/config/ur3_calibration.yaml [script_sender_port:=SCRIPT_SENDER_PORT]
'
but I got a same error.

I'm thinking running externalcontrol program properly in teaching pendant and setting custom port properly may solve this problem, but I don't know how.

## Expected Behavior
connection between ROS pc and UR3 will be establshed.

## Actual Behavior
ur3_bringup.launch is terminated.

7519060, 2022-12-20:
I'm so sorry for not previewing the issue before I post. I didn't know I could preview. It seems something went wrong when I paste error messages and codes.

hantong96, 2022-12-22:
I had the same problem.
ur10 3.10.1
ubuntu18.04 melodic
externalcontrol-1.0.5.urcap

fmauch, 2023-01-03:
Hi thank you for reporting this. This is definitively a regression introduced in #566.

fmauch, 2023-01-03:
Recalling on this. As #566 states, the minimum required Polyscope version would be 3.12.0. @hantong96 you use 3.10.1. @7519060 you use 3.8.0.

We need to update our documentation.

7519060, 2023-01-04:
@fmauch 
Thank you for your reply.
I updated my polyscope version to 3.15.8 and it worked!

closed 2023-01-04 by 7519060