UniversalRobots/Universal_Robots_ROS_Driver #607: Robot failed to receive joint/io states after UR Computer reboot

shuobh, 2023-01-17:
# Summary
*Introduction to the issue*
The issue is related to https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/272.
We run into situations that the UR Computer freezes and would need a reboot to recover. However, after UR computer reboots, we can connect to the dashboard using `rosservice call /ur_hardware_interface/dashboard/connect "{}"` and do power on/brake release/play/... but the robot would no longer publish it's state such as joint states/io states/wrench.

# Versions
 - ROS Driver version: 2.0.0
 - Affected Robot Software Version(s): 5.11.1
 - Affected Robot Hardware Version(s): UR3e

# Impact
Currently to recover from a UR Computer reboot, we have to restart all ROS nodes, which will lose the current state of the system.

## Steps to Reproduce
1. Run ROS Driver
2. Reboot UR computer
3. Monitor UR IO by `rostopic echo /ur_hardware_interface/io_states`

fmauch, 2023-01-18:
Hmm, I think this should be working, but it obviously does not.

shuobh, 2024-06-14:
One way to resolve this without impacting other node is to change all node launched by this package to `respawn="true"` and kill all of them with `rosnode kill /robot_state_publisher & rosnode kill /ros_control_controller_spawner & rosnode kill /ros_control_stopped_spawner & rosnode kill /controller_stopper & rosnode kill /ur_hardware_interface/ur_robot_state_helper & rosnode kill /ur_hardware_interface`

closed 2024-06-14 by shuobh