UniversalRobots/Universal_Robots_ROS_Driver #608: controller failed during execution

rijalmadhav12, 2023-01-19:
# Summary

*Introduction to the issue*
I am using external control urcap to control the real UR5 robot. I can control the robot from test_move program and rqt_joint _trajectory controller as well. But i am not being able to execute the plan from Move it, as it says ABORTED: Solution found but controller failed during execution

# Versions
 - ROS Driver version:Ros noetic
 - Affected Robot Software Version(s):cb3
 - Affected Robot Hardware Version(s):ur5
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):3.12

# Issue details
*This part is important in order to clarify the severity of the issue.*
**In terminal 1** :I run roslaunch ur_robot_driver ur5_bringup.launch robot_ip:=192.168.10.111, this works perfectly.
**In terminal 2** : roslaunch ur5_moveit_config moveit_planning_execution.launch
the error message are
[ INFO] [1674155962.184395478]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1674155962.184402902]: Using planning request adapter 'Fix Start State Path Constraints'
[ WARN] [1674155967.195404824]: Waiting for /scaled_pos_traj_controller/follow_joint_trajectory/follow_joint_trajectory to come up
[ WARN] [1674155973.195700073]: Waiting for /scaled_pos_traj_controller/follow_joint_trajectory/follow_joint_trajectory to come up
[ERROR] [1674155979.195943555]: Action client not connected: /scaled_pos_traj_controller/follow_joint_trajectory/follow_joint_trajectory
[ INFO] [1674155979.213708782]: Returned 0 controllers in list
[ INFO] [1674155979.233806364]: Trajectory execution is managing controllers
[ INFO] [1674155979.233910378]: MoveGroup debug mode is OFF
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1674155979.315346090]: 

* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService

[ INFO] [1674155979.315713267]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1674155979.315736218]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1674156006.881511229]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1674156006.888197507]: Planner configuration 'manipulator[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1674156006.891181725]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1674156006.891298642]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1674156006.891559154]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1674156006.891751603]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1674156006.902767133]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1674156006.903632202]: manipulator/manipulator[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1674156006.903714888]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1674156006.903792268]: manipulator/manipulator[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1674156006.904043275]: ParallelPlan::solve(): Solution found by one or more threads in 0.014201 seconds
[ INFO] [1674156006.904352470]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1674156006.904449511]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[... 120 lines cut ...]
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase
  - controller_interface::ControllerBase

## Expected Behavior

I checked through previous similar issue( issue #55 ). Modifying the controller.yaml should solve the issue. But it has not been solved in my case.

fmauch, 2023-01-27:
Your `ros_controller.yaml` file seems wrong. Where did you get that from? If that's from somewhere in our repositories, we'll have to update this.

The `scaled_pos_traj_controller` has been renamed to `scaled_pos_joint_traj_controller` in the past (#214). So, your file should like [the one from the support repo](https://github.com/ros-industrial/universal_robot/blob/melodic-devel/ur5_moveit_config/config/ros_controllers.yaml).

rijalmadhav12, 2023-01-31:
it is from your repo. thank you very much. it solved my issue.

fmauch, 2023-02-03:
> it is from your repo. thank you very much. it solved my issue.

What do you mean with that? https://github.com/fmauch/universal_robot/tree/calibration_devel?

That should not be used anymore and hopefully isn't mentioned anywhere in our docs, anymore.

rijalmadhav12, 2023-02-03:
https://github.com/ros-industrial/universal_robot.git src/universal_robot ,the moveit config is coming from this.

closed 2023-06-04 by github-actions[bot]