UniversalRobots/Universal_Robots_ROS_Driver #611: The way to control Onrobot RG2 gripper (UR5e and QC-R v3 Qiuck Changer)

pkowalsk1, 2023-01-23:
# Summary

I am trying to control the Onrobot RG2 gripper on the UR5e. It is attached to the robot using the [QC-R v3](https://onrobot.com/en/products/quick-changer) Qiuck Changer so it is possible to control the operation of the gripper with the eight-pinned connector (adjacent to the tool flange on Wrist #3). URcap was supplied with the gripper, so it can be easily controlled from the UR GUI and URScript. I can also control the ur5e with ROS driver and Moveit! 

1. As far as I know, if I want to control the gripper with ROS, I should either use the ROS driver provided by the manufacturer (which I couldn't find) or publish the URScript on topic `/ur_hardware_interface/script_command`. Are there other methods or have I missed something?

2. I found ROS packages that use Modbus/TCP (i.e [onrobot](https://github.com/Osaka-University-Harada-Laboratory/onrobot)) but it's **not** possible to use them with this RS485/IO connection, right?

3.  I tried to publish `set_digital_out(0, True)` at `/ur_hardware_interface/script_command` topic and it was successful. It is possible to use `rg_grip()` function in URscript and it works very well, however when I publish this function it has no effect. I tried:

- publish it as normal and secondary function:
```bash
rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'rg_grip(100.0, 40.0, tool_index = 0, blocking = True, depth_comp = False, popupmsg = True)'"
```
```bash
rostopic pub /ur_hardware_interface/script_command std_msgs/String "data: 'sec myProg();
    rg_grip(100.0, 40.0, tool_index = 0, blocking = True, depth_comp = False, popupmsg = True)
end'"
# also tested with 'def' instead of 'sec'
```
I even wrote publisher to make sure the script was formatted correctly.

- I send URscript generated in UR GUI (1.5k lines) via socket with the same result ([link](http://axisnj.com/controlling-a-universal-robots-cobot-using-python/)).

- I tried also load URscript (which worked run in GUI) with [load_program](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#load_program-ur_dashboard_msgsload) service but without success. Have I missed something important? Can I suspect that the `rg_grip()` function is time management and therefore cannot be run as a secondary function? It also cannot be run as a regular function because the manipulator control program is already running behind it.

How do you think I can control the manipulator in the best way?

# Versions
 - ROS Driver version: 2.1.0
 - Affected Robot Software Version(s): URSoftware 5.12.4.1101661 (Aug 31 2022)
 - Affected Robot Hardware Version(s): n/a
 - Robot Serial Number: 2022559--83
 - UR+ product(s) installed: n/a
 - URCaps Software version(s): OnRobot v5.18.0

# Issue details
I launched  `ur_robot_driver/launch/ur5e_bringup.launch` with this configuration:
```xml
<include file="$(find ur_robot_driver)/launch/ur5e_bringup.launch">
  <arg name="robot_ip" value="$(arg robot_ip)" />
  <arg name="tf_prefix" value="$(arg tf_prefix)" />
  <arg name="controller_config_file" value="$(arg controller_config_file)" />
  <arg name="robot_description_file" value="$(arg robot_description_file)" />
  <arg name="kinematics_config" value="$(arg kinematics_config)" />
  <arg name="use_tool_communication" value="True" />
  <arg name="tool_device_name" value="/dev/ttyTool" />
  <arg name="tool_voltage" value="24" />
  <arg name="tool_baud_rate" value="1000000" />
  <arg name="tool_rx_idle_chars" value="2.0" />
  <arg name="tool_tx_idle_chars" value="4.0" />
  <arg name="headless_mode" value="true" />
</include>
```

## Use Case and Setup
UR5e with Onrobot RG2 gripper mounted on [Panther](https://husarion.com/manuals/panther/) robot. Griper is attached to the robot arm using the [QC-R v3](https://onrobot.com/en/products/quick-changer) Qiuck Changer so it is possible to control the operation of the gripper with the eight-pinned connector (adjacent to the tool flange on Wrist #3).

fmauch, 2023-01-27:
When using an e-Series robot the robot has to be in remote_control mode in order to accept custom script commands. 

As a workaround you can execute some script when an output is set. I don't quite remember, but out of my head a TP program such as this could work:

[image]

Basically, in a separate thread you wait for an input to be set and then execute some script code. It's a workaround, but it can work rather reliable.

Alternatively, since you seem to be familiar with URScript already, you could extend the relatively new [script command interface](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/control/script_command_interface.cpp) to inject a custom command. See the corresponding [commit](https://github.com/UniversalRobots/Universal_Robots_Client_Library/commit/ba604f75f013f16b1942f3e207835b05fac77f78).

ericahellscythe, 2023-02-08:
Hi have you managed to get control of the OnRobot tools remotely using your computer? I am in the exact same situation as you. Writing a TP program to run a script works but that is not what I want. I tried to run a python program to control the RG2 but it failed. But if I were to run the program to control other stuff (i.e changing of I/O or moving of joints), it works. 

```
import socket
import time

HOST = "192.168.10.11"
PORT = 30002

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect((HOST, PORT))

#====================== Works =======================

s.send(("set_digital_out(0, True)"+"\n").encode('utf8'))
time.sleep(1)
data = s.recv(1024)

#==================================================

# ============= OnRobot Script does not work =============

f = open("/home/zhy/ur10_project/src/simple_program/scripts/test.script", "rb")
l = f.read(1024)

while (l):
    s.send(l)
    l = f.read(1024)

#==================================================
```

My test.script:
`rg_grip(rg_Width+30, 20, rg_index_get())`

Is there a way to control OnRobot products without using the compute box? Any help is appreciated.

ericahellscythe, 2023-02-09:
So I created the same UR program as you. But when I called the rosservice:

> rosservice call /ur_hardware_interface/set_io "fun: 1
> pin: 0
> state: 1.0" 

> 
> The teach pendent gives me an error:
> RG Grip command returned with an error. Please be sure that the required width is between the limits.
> Program halted.

Which line in the UR program am I suppose to edit? Is it line 5?
How do I tell what's the required width to change to?
And may I check with you if your last part of your line 8 script is just tool_index=0 and the other insignificant arguments?

Thanks

pkowalsk1, 2023-03-29:
Hi,

Sorry for the delay, but we now have the latest stable and tested version of the ROS 2 driver for the OnRobot RG2 gripper, which uses the MODBUS protocol. You can find the Docker configuration for this gripper in [ur-onrobot-rg2-docker](https://github.com/husarion/ur-onrobot-rg2-docker) repository, and the integration with the UR5e Docker configuration in [ur-docker](https://github.com/husarion/ur-docker) repo. This allows you to launch ROS 2 nodes from the [ur_onrobot_rg2_ros](https://github.com/husarion/ur_onrobot_rg2_ros) repository.

Regarding the issue you mentioned with width limits, I encountered a similar problem in the past when the tool ID was incorrect. However, all the grippers in our office had the same ID, which was set to 0.