UniversalRobots/Universal_Robots_ROS_Driver #616: Remotely forcing a UR robot into reduced mode?

ThisIsCertainlyNotAfakeAccount, 2023-02-07:
*Question*
I am trying to force my UR5e into reduced mode remotely. I looked into the URScript manual and the RTDE but I can seem to find a way to do so. Am I missing something or is this not possible?

My goal would be to implement this as a service.

Any help is appreciated!
```
ROS Driver version: ROS Melodic
Robot Software Version(s): 5.13
Robot Hardware Version(s): UR5e
```

RobertWilbrandt, 2023-02-08:
I am not aware of any way to get into reduced mode that doesn't involve safety I/Os, installation features or PROFIsafe. The only interaction with it through this driver are through the safety state published on `/ur_hardware_interface/safety_mode` which tells you if you are currently in normal or reduced mode. I am also not convinced that what you are asking for is possible, as going into a safety mode from a not-safety-certified device doesn't really help you in any safety critical setup. You might get a more definitive answer to this in the [UR forum](https://forum.universal-robots.com/) though, as the ability to get into this mode is not specific to the ROS driver.

If you don't actually need to go into reduced mode but are looking for a way to reduce motion speeds you might be interested in setting the speed scaling instead. It is published on `/speed_scaling_factor` and can be set through the `/ur_hardware_interface/set_speed_slider` service.

ThisIsCertainlyNotAfakeAccount, 2023-02-08:
ok thank you

closed 2023-02-08 by ThisIsCertainlyNotAfakeAccount