UniversalRobots/Universal_Robots_ROS_Driver #617: Wrench topic publishes significant force values when no force is being applied to the UR10e

SCerq, 2023-02-08:
# Summary
Hey there!
I am using the topic /Wrench to get the force and torque reading from the  FT sensor included in the UR10e.
I was expecting to get values near 0 when the robot is stopped and there is no interaction force on the robot's end-effector. 
However,  instead I am getting some considerable values that change with the robot position. 
I read somewhere that the CB3 series has an internal force sensor whose measured forces I imagine are derived from the joint currents. Therefore, it would make sense that the force sensed on the robots gripper changed with the robots configuration since the current changes according to the gravity compensation. 

Summing up, Could this topic be publishing the internal force sensor and not the force sensed on the FT sensor?

If not, how can I "calibrate" the sensor? For my application I need an accurate  force measurement.
 

*Introduction to the issue*

# Versions
 - ROS Driver version: 
 - Affected Robot Software Version(s): 5.11.8 (Jan 2022)
 - Affected Robot Hardware Version(s): UR10e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s): 1.0.5
[image]

# Impact
I can't continue my controller implementation with this force values. I am sending velocity commands to the robot that derive from the measured force.

## Steps to Reproduce
*Make simple example to reproduce the issue. Try to remove dependencies to other hardware and software components, if it is possible.*

## Expected Behavior
I expected to receive a wrench ={0,0,0,0,0,0} when there was no force being applied to the robot. 

## Actual Behavior
Attached to this is an image with the values published by the wrench topic. The wrench topic publishes considerable force values that change according to the robot configuration.  At first glance, might suggest that the measured force is influenced by gravity.
[image]

Thanks in advance!!!

gavanderhoorn, 2023-02-09:
> I expected to receive a wrench ={0,0,0,0,0,0} when there was no force being applied to the robot.

while I believe this would be difficult (fi, due to noise), things should get better after calling the [zero_ftsensor](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/395c0541b20d0da2cd480e2ad85b2100410fb043/ur_robot_driver/doc/ROS_INTERFACE.md#zero_ftsensor-std_srvstrigger) service.

urrsk, 2023-02-10:
The Force-Torque sensor is a relative sensor placed in the robot tool flange on the E-Series. You zero it using the [zero_ftsensor](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/395c0541b20d0da2cd480e2ad85b2100410fb043/ur_robot_driver/doc/ROS_INTERFACE.md#zero_ftsensor-std_srvstrigger) service as @gavanderhoorn suggested or by the zero_ftsensor() URScript command.

A normal way to use the FT sensor is to
1. Stand still in a short while to avoid vibrations
2. Zero the sensor
3. Use the sensor values...

SCerq, 2023-02-13:
Thanks @gavanderhoorn  and @urrsk for your suggestions!

I experimented with the zero_ftsensor service as you both suggested. I called the service using the command line. Even though I received a success response, the offsets did not changed much. Do you have any idea why?

Thank you both again!

urrsk, 2023-02-15:
@SCerq what version of the UR driver are you using?
We have changed the way the zero_ftsensor service work on version 2.1.0, see this [commit](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/c10624426f3fbc55986add95bc41f509ef4650bd)

closed 2023-06-16 by github-actions[bot]