UniversalRobots/Universal_Robots_ROS_Driver #624: How to switch the default controller from scaled_pos_joint_traj_controller to joint_group_vel_controller

stevensu1838, 2023-03-25:
Can any of you tell me which file I need to modify to switch the default controller from scaled_pos_joint_traj_controller to joint_group_vel_controller? 

It doesn't look like it is the ur5e_bringup.launch below:

[image]

stefanscherzinger, 2023-03-25:
Hi @stevensu1838 

I think changing your line `10` to
```xml
<arg name="controllers" default="joint_state_controller joint_group_vel_controller speed_scaling_state_controller force_torque_sensor_controller" doc="Controllers that are activated by default."/>
```
should do the job.
Are you working on the latest version? Your file seems to be missing `<arg name="script_command_port" ` [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blame/master/ur_robot_driver/launch/ur5e_bringup.launch#L9) that was touched 4 months ago. (just asking..)

stevensu1838, 2023-03-26:
@stefanscherzinger Hi, thanks a lot for your reply. You are right. my file is missing <arg name="script_command_port". But I followed the official documents to install this ROS UR robot driver a few weeks ago. Do I need to update to the latest version?

And apart from the joint_group_vel_controller, my I please ask what the other controllers do in this line? when do they work?

<arg name="controllers" default="joint_state_controller joint_group_vel_controller speed_scaling_state_controller force_torque_sensor_controller" doc="Controllers that are activated by default."/>

You can find my complete file down here. Thank you so much again. 
<?xml version="1.0"?>
<launch>
  <arg name="debug" default="false" doc="Debug flag that will get passed on to ur_common.launch"/>
  <arg name="robot_ip" doc="IP address by which the robot can be reached."/>
  <arg name="reverse_ip" default="" doc="IP of the driver, if set to empty it will detect it automatically."/>
  <arg name="reverse_port" default="50001" doc="Port that will be opened by the driver to allow direct communication between the driver and the robot controller."/>
  <arg name="script_sender_port" default="50002" doc="The driver will offer an interface to receive the program's URScript on this port. If the robot cannot connect to this port, `External Control` will stop immediately."/>
  <arg name="trajectory_port" default="50003" doc="Port that will be opened by the driver to allow trajectory forwarding."/>
  <arg name="tf_prefix" default="" doc="tf_prefix used for the robot."/>
  <arg name="controllers" default="joint_state_controller joint_group_vel_controller speed_scaling_state_controller force_torque_sensor_controller" doc="Controllers that are activated by default."/>
  <arg name="stopped_controllers" default="pos_joint_traj_controller joint_group_vel_controller" doc="Controllers that are initally loaded, but not started."/>
  <arg name="controller_config_file" default="$(find ur_robot_driver)/config/ur5e_controllers.yaml" doc="Config file used for defining the ROS-Control controllers."/>
  <arg name="robot_description_file" default="$(find ur_description)/launch/load_ur5e.launch" doc="Robot description launch file."/>
  <arg name="kinematics_config" default="$(find ur_description)/config/ur5e/default_kinematics.yaml" doc="Kinematics config file used for calibration correction. This will be used to verify the robot's calibration is matching the robot_description."/>
  <arg name="use_tool_communication" default="false" doc="On e-Series robots tool communication can be enabled with this argument"/>
  <arg name="tool_voltage" default="0" doc="Tool voltage set at the beginning of the UR program. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_parity" default="0" doc="Parity configuration used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_baud_rate" default="115200" doc="Baud rate used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_stop_bits" default="1" doc="Number of stop bits used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_rx_idle_chars" default="1.5" doc="Number of idle chars in RX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_tx_idle_chars" default="3.5" doc="Number of idle chars in TX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_device_name" default="/tmp/ttyUR" doc="Local device name used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_tcp_port" default="54321" doc="Port on which the robot controller publishes the tool comm interface. Only used, when `use_tool_communication` is set to true."/>
  <arg name="headless_mode" default="false" doc="Automatically send URScript to robot to execute. On e-Series this does require the robot to be in 'remote-control' mode. With this, the URCap is not needed on the robot."/>
  <arg name="ur_hardware_interface_node_required" default="true" doc="Shut down ros environment if ur_hardware_interface-node dies."/>

  <include file="$(find ur_robot_driver)/launch/ur_common.launch" pass_all_args="true" />
</launch>

stevensu1838, 2023-03-26:
@stefanscherzinger Hi, after I make the change as you suggested, I get the following errors when I run roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.177. Can you please help?

[INFO] [1679860326.835185]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1679860326.839693]: Loading controller: pos_joint_traj_controller
[INFO] [1679860326.850323]: Loading controller: joint_state_controller
[INFO] [1679860326.884505]: Loading controller: joint_group_vel_controller
[INFO] [1679860326.888433]: Loading controller: joint_group_vel_controller
**[ERROR] [1679860326.892252783]: A controller named 'joint_group_vel_controller' was already loaded inside the controller manager**
[INFO] [1679860326.892432]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
**[ERROR] [1679860327.893750]: Failed to load joint_group_vel_controller**
[INFO] [1679860327.897184]: Loading controller: speed_scaling_state_controller
[INFO] [1679860327.913082]: Loading controller: force_torque_sensor_controller
[INFO] [1679860327.929072]: Controller Spawner: Loaded controllers: joint_state_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1679860327.939049]: Started controllers: joint_state_controller, speed_scaling_state_controller, force_torque_sensor_controller

closed 2023-03-26 by stevensu1838