UniversalRobots/Universal_Robots_ROS_Driver #625: Is it possible using xenomai for rtde?

seungjaemoon, 2023-03-25:
Hi, i want to use ur-rtde and ur robot driver.

Now, i'm using xenomai 3 and 5.4.102-xenomai-3.1 kernel.

Is it possible using xenomai instead PREEMPT_RT for rtde?

seungjaemoon, 2023-03-26:
And I have one more question.

what's the difference between using servoj with ur robot driver and direct use of servoj with ur rtde?

fmauch, 2023-07-03:
Sorry for not answering this so far. I am not familiar with xenomai and therefore cannot give a qualified answer.

Regarding the second question: The difference lies in the underlying communication structure and the actual application interface. Which one fits your application better, you'll have to decide.

gavanderhoorn, 2023-07-03:
Xenomai is "just another way" to achieve determinism on a Linux system (don't tell the Xenomai devs). It's a co-kernel approach, which arguably achieves lower latencies and jitter than a PREEMPT_RT.

Similar care as with PREEMPT_RT would have to be taken to not break determinism, but it should be possible to use it with `ur_robot_driver` in a similar way.

(no guarantees though as Xenomai tends (or tended to be) a little bit more picky when it comes to which APIs are allowed in an RT context)

fmauch, 2024-07-03:
I don't think I'll have anything to add to what has been written. Therefore, closing this.

closed 2024-07-03 by fmauch