UniversalRobots/Universal_Robots_ROS_Driver #631: CMake Error: The following variables are used in this project, but they are set to NOTFOUND.

pkoprov, 2023-04-29:
Thanks @FinnLinxxx for trying the suggestions and your detailed output. It's probably two issues. The first one being the obscure `/include` whose origin we need to clarify and the second being the unset `Boost_INCLUDE_DIR` that cmake is complaining about.

Concerning the second issue, from your build output, I get that Boost is found via 

> /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake

which only defines `Boost_INCLUDE_DIRS` via `get_target_property(Boost_INCLUDE_DIRS Boost::headers INTERFACE_INCLUDE_DIRECTORIES)` but not `Boost_INCLUDE_DIR`. Note the additional `S`.

I'm not sure right now, but I could imagine that [this line ](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_kinematics/CMakeLists.txt#L26)dates back to older Boost versions.
Inside your `fmauch_universal_robot/ur_kinematics/CMakeLists.txt`, try replacing the include (line 26) with this:

```yaml
include_directories(SYSTEM ${Boost_INCLUDE_DIRS})
```

to see if this fixes the build issue.

_Originally posted by @stefanscherzinger in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/432#issuecomment-895387436_

pkoprov, 2023-04-29:
Same issue is persistent. When the line 26 in src/universal_robot/ur_kinematics/CMakeLists.txt contains include_directories(SYSTEM ${Boost_INCLUDE_DIR}) instead of `include_directories(SYSTEM ${Boost_INCLUDE_DIRS})

pkoprov, 2023-05-02:
Update:
The issue does not persist with Nvidia Jetson Xavier DevKit on Jetpack 5.1.1 Tegra R35.3.1 "20.04.6 LTS (Focal Fossa)"
The package builds without problems, even though line 26 in CMakeListsts.txt of ur_kinematics package contains  `include_directories(SYSTEM ${Boost_INCLUDE_DIR})`

JKH-ConeTec, 2024-03-18:
Copying my comment over from Issue https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/432#issuecomment-2005011899:

> I have noticed that if I  `catkin_make` right from the computer it fully compiles and the errors are absent.  However, if I `catkin_make` from a remote connection such as SSH or RDP, then both of these errors arise.
> 
> **Solution:**
> 
> - If possible, try and `catkin_make` right from the computer, and _not_ from a remote connection such as SSH or RDP.
> 
> - If method accessing the computer directly is not possible for your situation, then running the following commands in your `catkin_ws` root will fix or "bypass" these errors:
>   ```
>   sudo mkdir /include
>   sed -i "s/Boost_INCLUDE_DIR/Boost_INCLUDE_DIRS/" src/universal_robot/ur_kinematics/CMakeLists.txt
>   ```

RobertWilbrandt, 2024-03-21:
We have now fixed the incorrect cmake variable in https://github.com/ros-industrial/universal_robot/pull/675. Please let us know if this fixes your issue.

closed 2024-04-17 by fmauch