UniversalRobots/Universal_Robots_ROS_Driver #633: How to get the raw data of force_torque_sensor?

KentoYamamotoNU, 2023-05-11:
I am using a robot hand (2F-85) from robotiq attached to UR. When UR is started, the value of force_torque_sensor is initialized and the force shown in the /wrench topic. 
wrench: 
  force: 
    x: -0.46766315205091247
    y: -0.05913990644121419
    z: 0.06575835894085908
  torque: 
    x: -0.00253918177657238
    y: 0.0173421697176658
    z: 0.01970360803376669
The posture at this time is the picture below.
[image]
The value of force_sensor is corrected and the value changes about 5 minutes later, even though UR is not moved.
wrench: 
  force: 
    x: -0.20049144606896044
    y: 1.3249557663686649
    z: 4.0906257492416165
  torque: 
    x: 0.003108101780185593
    y: 0.009844995134569583
    z: -0.03750110083972966
Also, if the robot hand was rotated 90 degrees around the x-axis and the y and z values will be different, as shown in the picture below.
wrench: 
  force: 
    x: -1.989279264783352
    y: 18.50540543168785
    z: 22.141448526040982
  torque: 
    x: 0.37064158239178424
    y: -0.03177618098109141
    z: 0.38942961673354903
    
[image]

How can I get the raw data of force_torque_sensor without any correction?

fmauch, 2023-05-15:
Hi,

> the value changes about 5 minutes later, even though UR is not moved.

Yes, the internal sensor drifts over time, see #294. Hence, to some extent this is exptected.

> Also, if the robot hand was rotated 90 degrees around the x-axis and the y and z values will be different

This is rather normal, as the wrench is "stamped", meaning it is assigned to a frame, in this case the `tool0_controller` frame. Since that rotates with the TCP, forces in that frame change, as well (since the force induced by something stiff mounted to the TCP is constant w.r.t. the global world frame.

```
header:
  seq: 4412
  stamp:
    secs: 1684139821
    nsecs: 805363098
  frame_id: "tool0_controller"
wrench:
  force:
    x: 0.0
    y: 0.0
    z: 0.0
  torque:
    x: 0.0
    y: 0.0
    z: 0.0
```

> The value of force_sensor is corrected 

I'm not completely sure what you mean by that. The FTS measurements from the robot are transformed into the TCP's frame (if required - depending on the robot's version) and otherwise not altered by this driver.

closed 2024-04-17 by fmauch

Fasteroid, 2024-05-02:
I don't understand how the force data is meant to be useful if it drifts over time.  Why was this closed?  I'm experiencing the same issue.