UniversalRobots/Universal_Robots_ROS_Driver #635: catkin_make failed!

LexTran, 2023-06-12:
# Summary
I was building UR kit following the instructions in README, however, I have encountered a problem when performing catkin_make.
The complete console message is too long, so I just put the tail of it here:

```
[ 98%] Building CXX object Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o
In file included from /usr/local/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/local/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/local/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/noetic/include/ros/forwards.h:37,
                 from /opt/ros/noetic/include/ros/common.h:37,
                 from /opt/ros/noetic/include/ros/ros.h:43,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration.h:32,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration.cpp:29:
/usr/local/include/boost/bind.hpp:36:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/local/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/local/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/noetic/include/ros/forwards.h:37,
                 from /opt/ros/noetic/include/ros/common.h:37,
                 from /opt/ros/noetic/include/ros/ros.h:43,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration.h:32,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration_consumer.h:35,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_consumer.cpp:29:
/usr/local/include/boost/bind.hpp:36:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/boost/smart_ptr/detail/sp_thread_sleep.hpp:22,
                 from /usr/local/include/boost/smart_ptr/detail/yield_k.hpp:23,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock_gcc_atomic.hpp:14,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock.hpp:42,
                 from /usr/local/include/boost/smart_ptr/detail/spinlock_pool.hpp:25,
                 from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:29,
                 from /usr/local/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/noetic/include/ros/forwards.h:37,
                 from /opt/ros/noetic/include/ros/common.h:37,
                 from /opt/ros/noetic/include/ros/ros.h:43,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration.h:32,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration_consumer.h:35,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:29:
/usr/local/include/boost/bind.hpp:36:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp: In member function ‘bool CalibrationCorrection::writeCalibrationData()’:
/home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:111:54: warning: ‘boost::filesystem::path boost::filesystem::complete(const boost::filesystem::path&)’ is deprecated: Use absolute() instead [-Wdeprecated-declarations]
  111 |     fs::path out_path = fs::complete(output_filename_);
      |                                                      ^
In file included from /usr/local/include/boost/filesystem.hpp:19,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:44:
/usr/local/include/boost/filesystem/operations.hpp:353:13: note: declared here
  353 | inline path complete(path const& p)
      |             ^~~~~~~~
/home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:111:54: warning: ‘boost::filesystem::path boost::filesystem::complete(const boost::filesystem::path&)’ is deprecated: Use absolute() instead [-Wdeprecated-declarations]
  111 |     fs::path out_path = fs::complete(output_filename_);
      |                                                      ^
In file included from /usr/local/include/boost/filesystem.hpp:19,
                 from /home/wxy/catkin_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:44:
/usr/local/include/boost/filesystem/operations.hpp:353:13: note: declared here
  353 | inline path complete(path const& p)
      |             ^~~~~~~~
[100%] Linking CXX executable /home/wxy/catkin_ws/devel/lib/ur_calibration/calibration_correction
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o: in function `boost::filesystem::path::parent_path() const':
calibration_correction.cpp:(.text._ZNK5boost10filesystem4path11parent_pathEv[_ZNK5boost10filesystem4path11parent_pathEv]+0x2c): undefined reference to `boost::filesystem::path::find_parent_path_size() const'
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o: in function `boost::filesystem::absolute(boost::filesystem::path const&, boost::filesystem::path const&)':
calibration_correction.cpp:(.text._ZN5boost10filesystem8absoluteERKNS0_4pathES3_[_ZN5boost10filesystem8absoluteERKNS0_4pathES3_]+0x3c): undefined reference to `boost::filesystem::detail::absolute(boost::filesystem::path const&, boost::filesystem::path const&, boost::system::error_code*)'
collect2: error: ld returned 1 exit status
make[2]: *** [Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/build.make:168：/home/wxy/catkin_ws/devel/lib/ur_calibration/calibration_correction] error 1
make[1]: *** [CMakeFiles/Makefile2:8728：Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/all] error 2
make: *** [Makefile:141：all] error 2
Invoking "make -j8 -l8" failed

```

# Versions
 - ROS Driver version: ROS1 Noetic
- OS version: Ubuntu 20.04 LTS

# Impact
Cannot build the repository, so I can not use UR kits to do Robot related research.

## Project status at point of discovered
In first time trying setup the needed environment.

## Steps to Reproduce
I strictly follow the steps in README, but I just failed to build

Anthony-EEE, 2023-10-11:
@LexTran I also got the error. DId you find any solution?

fmauch, 2023-10-12:
Sorry, that you didn't get any answer so far.

Your output looks as if you have installed your own version of boost, since headers in `/usr/local/include` are used instead of the headers provided by the ubuntu packages in `/usr/include`. When you've installed your own boost version without the filesystem module, you might run into the trouble experienced.

However, you could probably also use the binary version for installation, simply run `sudo apt install ros-noetic-ur-robot-driver`. We are currently updating the README about this, as this was missing so far...

FFF-xiaobai, 2024-03-04:
@LexTran   @Anthony-EEE   Hello, I have made the same mistake. Do you have any solution for it?

VinDp, 2024-05-22:
Since some time has passed from the proposed solution, I will close this issue. 
Feel free to re-open it if still relevant.

closed 2024-05-22 by VinDp