UniversalRobots/Universal_Robots_ROS_Driver #637: How to reconnect driver after "restart_safety" call

cedricgoubard, 2023-06-13:
# Summary

Calling the `restart_safety` service causes the driver to lose the connection with the robot, which forces me to restart the node. Is there a way around that?

# Versions
 - ROS Driver version: 2.0.0
 - Affected Robot Software Version(s): 5.9.3
 - Affected Robot Hardware Version(s): UE10e
 - URCaps Software version(s): External Control: 1.0.5

# Impact
The robot is unable to autonomously recover from collisions, I have to manually restart the driver.

# Issue details
Hello,
First, thank you for providing this driver!

I have been using it to control a UR10e for some time now, and everything works fine.
However, when the robot goes into emergency stop after a collision, the only solution I have found is to call the `/ur_hardware_interface/dashboard/restart_safety` service which does reset the robot, but also causes the driver connection to drop:

```log
[ WARN] [1686658080.478251539]: Failed to read from stream, reconnecting in 64 seconds...
[ERROR] [1686658080.491336357]: Could not get fresh data package from robot
[ERROR] [1686658080.591479936]: Could not get fresh data package from robot
[ERROR] [1686658080.691613176]: Could not get fresh data package from robot
[ERROR] [1686658080.791734523]: Could not get fresh data package from robot
...
```

I tried using the `connect` service, but it does not seem to change anything.
I am using a dedicated network, but could not use the RT kernel since I need the NVIDIA driver.

Things work again if I kill the node and restart it; but is there a way to regain control without having to resort to that?

fmauch, 2024-05-29:
Reconnections should work, so that's definitively a bug, **however** to solve your particular problem (if it still exists) there is an easier way:

You don't have to restart the safety system, but you can just acknowledge everything and continue using the robot state helper:

```
rostopic pub /ur_hardware_interface/set_mode/goal ur_dashboard_msgs/SetModeActionGoal "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
goal_id:
  stamp:
    secs: 0
    nsecs: 0
  id: ''
goal:
  target_robot_mode:
    mode: 7
  stop_program: true 
  play_program: true"
```

You probably have to quit the popup and restart the robot anyway to move out of the collision, right?

cedricgoubard, 2024-05-29:
Hi @fmauch,
Thank you for replying; I actually found another "solution" since asking: using a dedicated network interface between the arm and the robot solved my issue... which is what you've been recommending in other issues, I should have listened from the start 😄

closed 2024-05-29 by cedricgoubard