UniversalRobots/Universal_Robots_ROS_Driver #638: Frequent connection to reverse interface dropped and Connection attempt on port 50001 while maximum number of clients (1) is already connected.

YongyuLiu03, 2023-06-14:
# Configuration

 - OS: Ubuntu 20.04
 - ROS noetic
 - Affected Robot Software Version(s): 5.5.1.82186 (Sep 05 2019)
 - Affected Robot Hardware Version(s): UR10e
 - URCaps Software version(s): 1.0.5
 - Using PREEMT_RT kernel, robot wired and PC wireless connected to a router (same LAN). 

# Issue 

Hi, I'm using ur_robot_driver along with moveit! motion planning to operate ur10e robot. The procedure is:
 1. `roslaunch ur_robot_driver ur10e_bringup.launch`
 2. run the external control program on TP
 3. `roslaunch urr10e_moveit_config moveit_planning_execution.launch`
 4. run python scripts that plan and execute plans using moveit interface
 
The issue is that `Connection to reverse interface dropped` and reconnection frequently happen, which preempts the robot at the middle of an execution. Later the warning `Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.` occurs and the driver is not able to reconnect to the robot anymore. I can reconnect them If I manually end and restart the program running on robot, however the same issue repeats very soon.  

As suggested in previous related issues, I have tried setting up a realtime kernel and adding `ur_driver_->setKeepaliveCount(5); ` in `Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp` after `ur_driver_->registerTrajectoryDoneCallback(
      std::bind(&HardwareInterface::passthroughTrajectoryDoneCb, this, std::placeholders::_1));`, but the problem remains and it's worth noticing that the value of the keepalive variable on TP is still 1 after adding that line and drops to -2 after the connection dropped. 

Wireless connection of PC under the same LAN is preferred in the project and the network is set to be holding only communications of the robots in the lab, while the PC and the robot are the only active ones, so I guess there could be other issues affecting the connection. I'm also concerned about the clients on port 50001, what could cause the port to be occupied? 

Thanks in advance for people looking at this issue!

# Full log of ur10e_bringup.launch terminal

```bash
... logging to /home/ai4ce-lab/.ros/log/f2c7aa98-0ac1-11ee-bb7e-9fc1441ed0dd/roslaunch-ai4celab-Precision-3660-5095.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ai4celab-Precision-3660:45121/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 500
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
 * /joint_based_cartesian_traj_controller/type: position_controll...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
[... 303 lines cut ...]
[ INFO] [1686754328.345220781]: Robot requested program
[ INFO] [1686754328.345257919]: Sent program to robot
[ INFO] [1686754328.415676646]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1686754340.297537875]: Connection to reverse interface dropped.
[ INFO] [1686754340.299757792]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1686754340.437484644]: Connection to reverse interface dropped.
[ INFO] [1686754342.649537708]: Robot requested program
[ INFO] [1686754342.649572696]: Sent program to robot
[ INFO] [1686754342.713279433]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1686754355.067486705]: Connection to reverse interface dropped.
[ERROR] [1686754355.067516945]: Sending data through socket failed.
[ INFO] [1686754355.069351912]: Robot connected to reverse interface. Ready to receive control commands.
[ERROR] [1686754363.023654933]: Sending data through socket failed.
[ INFO] [1686754363.025435297]: Robot connected to reverse interface. Ready to receive control commands.
[ERROR] [1686754385.113368790]: Sending data through socket failed.
[ INFO] [1686754385.113407184]: Connection to reverse interface dropped.
[ INFO] [1686754385.115558491]: Robot connected to reverse interface. Ready to receive control commands.
[ERROR] [1686754386.717240783]: Sending data through socket failed.
[ INFO] [1686754386.717268209]: Connection to reverse interface dropped.
[ INFO] [1686754386.719053810]: Robot connected to reverse interface. Ready to receive control commands.
[ WARN] [1686754410.728921891]: Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.
[ERROR] [1686754410.811123082]: Sending data through socket failed.
[ INFO] [1686754410.811142606]: Connection to reverse interface dropped.
[ERROR] [1686754411.301704731]: Can't accept new action goals. Controller is not running.
```

shrutichakraborty, 2023-12-11:
Has anyone solved this issue? I am facing it as well ...

JohnL-Fxa, 2024-05-10:
I'm having a similar problem after upgrading from a ros melodic to noetic and from ubunto 18 to 20, I'm also using a raspberry pi instead of a laptop.
everything was working fine until I get the error of sending data through socket faild then the connection keeps dropping and reconnecting so the robot skips some commands because the connection is unstable till the controller stops running.

fmauch, 2024-05-15:
Regarding the OP: Please do not use a wireless connection.

@shrutichakraborty, @JohnL-Fxa Please open separate issues describing your setup.

closed 2024-05-15 by fmauch