UniversalRobots/Universal_Robots_ROS_Driver #639: How to use the cartesian trajectory controller?

devvaibhav455, 2023-06-16:
Hi,

I have a UR5 robot but before I test my stuff on the real robot, I want to make sure everything works fine in gazebo simulation. I want the robot’s EE to follow a 6-DOF pose continuously. I have got a 6-DOF joystick which publishes poses continuously and I want the robot's EE to follow it. I can do so easily using the franka emika cartesian impedance controller which they have already provided as an example and that makes me think that such a thing is possible.

For UR5, after [searching](https://forum.universal-robots.com/t/hallo-i-have-a-question-about-new-feature-cartesian-trajectory-controller-how-can-i-start-this-action-server-and-where-is-controller-yaml-file-which-i-need-to-modiefy/14115) I found out that pose_based_cartesian_traj_controller is something which I should be looking at. The answer requests us to go to beta-testing branch which then redirects to the staging branch

    This beta-testing branch is deprecated. The changes from this branch now live on the [staging](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/staging) branch.

I cloned the staging branch but unsure how to move forward. Could someone please provide step-by-step (if that's not possible, some guidance)  instructions to do that?

System details: Ubuntu 20.04 + ROS noetic on amd64 architecture
PS: I can switch to ROS melodic as well if that’s a hard-and-fast requirement.

Thanks in advance.

fmauch, 2024-05-29:
Well, my answer is a bit late, but nevertheless...

The Cartesian controllers have been merged and released quite some time ago, so using the released versions should be sufficient to use the Cartesian trajectory controllers.

However, for your use case a Cartesian trajectory controller might not be the sufficient one, since you (as far as I understood) do not want to execute predefined trajectories, but your want to make the EE follow an external pose source.

For that purpose you could use my colleague's https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers package. Those controllers exactly want an external pose topic as an input.

fmauch, 2024-07-03:
Closing this due to a proposed solution and no answer since then. Please feel free to comment / repoen if the issue still persists.

closed 2024-07-03 by fmauch