UniversalRobots/Universal_Robots_ROS_Driver #641: How to fix the orientation of end-effector during movement?

ZZWang21, 2023-07-06:
I have made a gazebo model of UR5 and ROBOTIQ 85 gripper. I would like to move the ur5 robot with the end-effector orientation fixed throughout the movemnt. is it possible? what should I use? I tried “constrainits” in moveit, it was not working.

UBUNTU 20.04, ROS NOETIC, UR5+ROBOTIQ 2F85 gripper. 

I expected the end effector to be fixed orientated vertical to the ground throughout the whole movement. but in fact, it was not moving that way. Could any one help me? Thanks. 

Original code: 

#!/usr/bin/env python

"""
    moveit_cartesian_path.py - Version 0.1 2016-07-28
    Based on the R. Patrick Goebel's moveit_cartesian_demo.py demo code.
    Plan and execute a Cartesian path for the end-effector.
    Created for the Pi Robot Project: http://www.pirobot.org
    Copyright (c) 2014 Patrick Goebel.  All rights reserved.
    This program is free software; you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation; either version 2 of the License, or
    (at your option) any later version.5
    This program is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details at:
    http://www.gnu.org/licenses/gpl.html
"""

import rospy, sys
import rospkg
import moveit_commander
from geometry_msgs.msg import Pose
from geometry_msgs.msg import PoseStamped
from moveit_msgs.msg import Constraints, OrientationConstraint
from copy import deepcopy
from std_msgs.msg import Header
from std_msgs.msg import String
from std_msgs.msg import Float64
from tf_conversions import posemath
from PyKDL import *
import tf
import math
#import forcezz

from trajectory_msgs.msg import JointTrajectory

from trajectory_msgs.msg import JointTrajectoryPoint
from geometry_msgs.msg import PoseStamped, Quaternion
import tf

import numpy as np

class MoveItCartesianPath:
    def __init__(self):
        rospy.init_node("moveit_cartesian_path", anonymous=False)

        rospy.loginfo("Starting node moveit_cartesian_path")

        rospy.on_shutdown(self.cleanup)

        # Initialize the move_group API
        moveit_commander.roscpp_initialize(sys.argv)

        # Initialize the move group for the ur5_arm
        self.arm = moveit_commander.MoveGroupCommander('arm')

        # Get the name of the end-effector link
        end_effector_link = self.arm.get_end_effector_link()
        print("ZZW END:",end_effector_link)

        # Set the reference frame for pose targets
        reference_frame = "base_link"

        # Set the ur5_arm reference frame accordingly
        self.arm.set_pose_reference_frame(reference_frame)

        # Allow replanning to increase the odds of a solution
        self.arm.allow_replanning(True)

        # Allow some leeway in position (meters) and orientation (radians)
        self.arm.set_goal_position_tolerance(0.01)  #ORI:0.01
        self.arm.set_goal_orientation_tolerance(0.01)

        # Get the current pose so we can add it as a waypoint
        start_pose = self.arm.get_current_pose(end_effector_link).pose

        # Initialize the waypoints list
        waypoints = []

        # Set the first waypoint to be the starting pose
        # Append the pose to the waypoints list
        waypoints.append(start_pose)
        print(waypoints)

        waypoints1 = start_pose
        # Define the target position (x, y, z) and the normal vector to the surface (nx, ny, nz)
        target_position = [0.3, 0.3, 0.15]
        surface_normal = [0, 0, 1]

        # Compute the quaternion corresponding to a rotation of 90 degrees around the X-axis
        #q = tf.transformations.quaternion_from_euler(0, 0, np.pi/4)
        # zzw 20230513 Create an orientation constraint object
        orientation_constraint = OrientationConstraint()
        orientation_constraint.header.frame_id = "base_link"
        orientation_constraint.link_name = end_effector_link
        orientation_constraint.orientation.x = start_pose.orientation.x#0.0
        orientation_constraint.orientation.y = start_pose.orientation.y
        orientation_constraint.orientation.z = start_pose.orientation.z
        orientation_constraint.orientation.w = start_pose.orientation.w
        orientation_constraint.absolute_x_axis_tolerance = 0.1
        orientation_constraint.absolute_y_axis_tolerance = 0.1
        orientation_constraint.absolute_z_axis_tolerance = 0.1
        orientation_constraint.weight = 1.0
        # Create a constraints object and add the orientation constraint
        constraints = Constraints()
        constraints.orientation_constraints.append(orientation_constraint)
        # Set the path constraints
        self.arm.set_path_constraints(constraints)

        # Create a PoseStamped message with the target position and orientation
        target_pose = PoseStamped()
        target_pose.header.frame_id = 'base_link'
        target_pose.pose.position.x = target_position[0]
        target_pose.pose.position.y = target_position[1]
        target_pose.pose.position.z = target_position[2]
        target_pose.pose.orientation = start_pose.orientation
       
        waypoints.append(target_pose)
        #(plan, fraction) = self.arm.compute_cartesian_path (waypoints, 0.001, 0.0, True)
        #self.arm.set_pose_target(target_pose, end_effector_link)
        
        self.arm.set_position_target(target_position, end_effector_link)
        #self.arm.set_pose_target(target_pose)
        #self.arm.set_max_velocity_scaling_factor(0.5) #ZZW SPEED
        # Plan and execute the trajectory to move the end effector to the target pose
        plan = self.arm.go(wait=True)
        trajectory_msg = JointTrajectory()
        for i in range(len(trajectory_msg.points)):
            print("Waypoint", i, "time stamp:", trajectory_msg.points[i].time_from_start.to_sec())

        # Clear the path constraints
        self.arm.clear_path_constraints()

        # Shutdown the moveit_commander module
        moveit_commander.roscpp_shutdown()
        #self.arm.stop()
        #self.arm.clear_pose_targets()

    def cleanup(self):
        rospy.loginfo("Stopping the robot")

        # Stop any current arm movement
        self.arm.stop()

        #Shut down MoveIt! cleanly
        rospy.loginfo("Shutting down Moveit!")
        moveit_commander.roscpp_shutdown()
        moveit_commander.os._exit(0)
# Force measurement 20220108:::: try class?

if __name__ == "__main__":
    try:
        MoveItCartesianPath()
    except KeyboardInterrupt:
        print ("Shutting down MoveItCartesianPath node.")

ZZWang21, 2023-07-07:
ALSO, after I move the rrobot, when I move the robot again, it showed error "

Joint 'shoulder_pan_joint' from the starting state is outside bounds by a significant margin: [ -416.721 ] should be in the range [ -6.28319 ], [ 6.28319 ] but the error above the ~start_state_max_bounds_error parameter (currently set to 0.1)
[ WARN] [1688701608.433455651, 5254.917000000]: Joint 'wrist_2_joint' from the starting state is outside bounds by a significant margin: [ -36.0934 ] should be in the range [ -6.28319 ], [ 6.28319 ] but the error above the ~start_state_max_bounds_error parameter (currently set to 0.1)
[ WARN] [1688701608.433565184, 5254.917000000]: Joint 'wrist_3_joint' from the starting state is outside bounds by a significant margin: [ -38.1583 ] should be in the range [ -6.28319 ], [ 6.28319 ] but the error above the ~start_state_max_bounds_error parameter (currently set to 0.1)"

I am not sure why. Thanks.

fmauch, 2023-07-07:
Having the robot keep a fixed orientation during a motion is something that needs to be handled on the path planning level, ergo moveit. As this is not a driver issue, I'm going to close this issue.

If you need help about moveit, I suggest finding it on https://robotics.stackexchange.com/

closed 2023-07-07 by fmauch