UniversalRobots/Universal_Robots_ROS_Driver #646: Connection setup failed

anbello, 2023-07-25:
# Summary

We are using your ROS driver with an UR3E. 
ROS noetic is installed on a NVIDIA Jetson Xavier with Ubuntu 20.04
The UR3E is connected to the Xavier via Ethernet.
The driver launch file is configured with headless_mode: True so we don't use URCaps Software.
With this settings we did a lot of testing and all worked w/o problems, then at a certain point the driver began to give the following error and then exit.

[ERROR] [1686134091.854418752]: Connection setup failed for 192.168.10.10:29999
[ INFO] [1686134091.892440224]: waitForService: Service [/ur_hardware_interface/dashboard/play] is now available.
[ INFO] [1686134091.919218464]: Initializing urdriver
[ERROR] [1686134094.926490304]: Connection setup failed for 192.168.10.10:30002
[ERROR] [1686134097.998459904]: Connection setup failed for 192.168.10.10:30004

# Versions
 - ROS Driver version: v2.0.0
 - Affected Robot Software Version(s):
 - Affected Robot Hardware Version(s): ur3e
 - Robot Serial Number:
 - UR+ product(s) installed:
 - URCaps Software version(s):

# Issue details
Here I post the complete log from the console after launching diver
```
urobo@ubuntu:~$ roslaunch ur_robot_driver ur3e_bringup.launch robot_ip:=192.168.10.10
... logging to /home/urobo/.ros/log/e27f6ed0-051e-11ee-9798-48b02d4d6d21/roslaunch-ubuntu-3493.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:33481/

SUMMARY
========

PARAMETERS
 * /controller_stopper/consistent_controllers: ['joint_state_con...
 * /force_torque_sensor_controller/publish_rate: 500
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 500
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
 * /joint_based_cartesian_traj_controller/type: position_controll...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 500
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 500
[... 217 lines cut ...]
[ERROR] [1686134091.854418752]: Connection setup failed for 192.168.10.10:29999
[ INFO] [1686134091.892440224]: waitForService: Service [/ur_hardware_interface/dashboard/play] is now available.
[ INFO] [1686134091.919218464]: Initializing urdriver
[ERROR] [1686134094.926490304]: Connection setup failed for 192.168.10.10:30002
[ERROR] [1686134097.998459904]: Connection setup failed for 192.168.10.10:30004
[FATAL] [1686134102.999718240]: Failed to connect to robot. Please check if the robot is booted and connected.
[ERROR] [1686134103.000051808]: Could not correctly initialize robot. Exiting
================================================================================REQUIRED process [ur_hardware_interface-2] has died!
process has died [pid 3511, exit code 1, cmd /home/urobo/catkin_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/urobo/.ros/log/e27f6ed0-051e-11ee-9798-48b02d4d6d21/ur_hardware_interface-2.log].
log file: /home/urobo/.ros/log/e27f6ed0-051e-11ee-9798-48b02d4d6d21/ur_hardware_interface-2*.log
Initiating shutdown!
================================================================================
[ur_hardware_interface/ur_robot_state_helper-6] killing on exit
[controller_stopper-5] killing on exit
[ros_control_stopped_spawner-4] killing on exit
[ros_control_controller_spawner-3] killing on exit
[ur_hardware_interface-2] killing on exit
[robot_state_publisher-1] killing on exit
[WARN] [1686134103.891691]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1686134103.894155]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_stopper-5] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
```

fmauch, 2023-08-02:
This usually appears if (as mentioned in the output) when the robot is either not connected or not booted.

You can check using `nmap -p29999-30004 192.168.10.10` whether the ports are actually open. For me with a non-connected robot that gives:

```
$ nmap -p29999-30004 192.168.56.101
Starting Nmap 7.80 ( https://nmap.org ) at 2023-08-02 17:12 CEST
Note: Host seems down. If it is really up, but blocking our ping probes, try -Pn
Nmap done: 1 IP address (0 hosts up) scanned in 3.06 seconds
```

With a connected, but not yet fully booted robot:
```
$ nmap -p29999-30004 192.168.56.101
Starting Nmap 7.80 ( https://nmap.org ) at 2023-08-02 17:12 CEST
Nmap scan report for 192.168.56.101
Host is up (0.00010s latency).

PORT      STATE  SERVICE
29999/tcp closed bingbang
30000/tcp closed ndmps
30001/tcp closed pago-services1
30002/tcp closed pago-services2
30003/tcp closed amicon-fpsu-ra
30004/tcp closed amicon-fpsu-s
```

And finally with a booted robot that I can start the driver with:
```
nmap -p29999-30004 192.168.56.101
Starting Nmap 7.80 ( https://nmap.org ) at 2023-08-02 17:13 CEST
Nmap scan report for 192.168.56.101
Host is up (0.00011s latency).

PORT      STATE  SERVICE
29999/tcp open   bingbang
30000/tcp closed ndmps
30001/tcp open   pago-services1
30002/tcp open   pago-services2
30003/tcp open   amicon-fpsu-ra
30004/tcp open   amicon-fpsu-s
```

If you don't see the `open` state for the relevant ports, please check your network setup. Of they show up as `open` and you still can't start the driver, please report back, then we need to do some further debugging.

fmauch, 2024-07-03:
Closing this due to inactivity. Please feel free to comment / reopen if the issue still persists.