UniversalRobots/Universal_Robots_ROS_Driver #647: Control RG2 gripper from ROS

masterdeki, 2023-07-28:
# Summary

Hi all.
I'm trying to control onrobot RG2 gripper from ROS, running python3 scripts from terminal but I have problems with connecting to gripper.

# Versions
 - ROS Driver version: 2.1.2
 - Affected Robot Software Version(s):
 - Affected Robot Hardware Version(s): UR5e
 - Robot Serial Number:
 - UR+ product(s) installed: 
 - URCaps Software version(s): 1.0.5

# Issue details
After I setup all things (like in use case and setup) and starting driver from terminal and externalcontrol from TP I can control robotic arm from moveit and rviz, and that works correctly. But now I need to also control gripper, possibly from moveit and rviz, but in worst case just from python script over terminal. I finded this [post](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/506#issuecomment-1256338704) and after running robot bringup command 'roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.0.50 kinematics_config:=$(rospack find ur_calibration)/etc/ur5e_calibration.yaml use_tool_communication:=true tool_device_name:=/tmp/ttyUR' everything is fine, but when I start external-control from pendant I get this error on TP 'Communication problem with an RG gripper. Please check the device' and this in terminal where bringup is started 
'''[ INFO] [1690540162.026777008]: Robot requested program
[ INFO] [1690540162.026839792]: Sent program to robot
[ INFO] [1690540162.288182102]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1690540162.938412505]: Connection to reverse interface dropped.'''
and after that on TP in installation>URCaps>OnRobot setup under device info Tool connector - Empty. 
If I run this 'rosrun onrobot_rg_control OnRobotRGTcpNode.py /tmp/ttyUR' after robot bringup I get this error 'Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.0.50:502)]'
In TP under settings for modbus I tryed to enter ip of the robot (192.168.0.50) and to add all 4 signals that are offered from dropdown menu (digital input, digital output, register input and register output). All 4 addresses are 0.

So, my question is should I run external-control from TP or no?And how to setup modbus correctly on TP? And where I'm making mistake?

I also must say that I'm new in all this and I just started to learning, and I hope that someone will help me.

## Use Case and Setup
My setup is ubuntu 22.04 laptop with ROS Noetic installed, connected to robo arm controller over ethernet cable. Robotic arm is UR5e, gripper is onrobot RG2 connected directly on arm flange with rs485. On TP I installed externalcontrol URcap, rs485 URcap and onrobot URcap from USB flash that comes with gripper. I followed tutorial from [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) and installed tool communication from [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/setup_tool_communication.md)
I also installed driver for gripper from [here](https://github.com/Osaka-University-Harada-Laboratory/onrobot)

## Expected Behavior
Expecting to run python script from terminal to open and close RG2 gripper atached to UR5e robotic arm.

fmauch, 2023-08-02:
Thank you for the intensive report. Given the information you posted, there seem to be a couple of things mixed up if I understand that correctly.

1. The [onrobot driver](https://github.com/Osaka-University-Harada-Laboratory/onrobot/tree/main) you linked seems to connect via ModbusTCP given an IP address, not through a tty device. I was a bit confused by the posted output of `Failed to connect[ModbusTcpClient(192.168.0.50:502)]`, which probably results from a previous attempt, so the parameter server contained the paramter `onrobot/ip` with value `92.168.0.50`. The given argument `/tmp/ttyUR` is not used by the `OnRobotRGTcpNode.py` node if I understand its source code correctly.
2. The tool forwarding only forwards the raw commands sent to the tty device. So, in order to use it with a modbus device, typically the ModbusRTU protocol is used.

One way forward might be to use the Onrobot URCap and setup digitial outputs to set predefined gripper positions from ROS.

masterdeki, 2023-08-10:
I want to use onrobot driver via modbus and given ip address, and control gripper from python script. What then should I do to procced?

closed 2023-10-31 by masterdeki