UniversalRobots/Universal_Robots_ROS_Driver #649: error: no matching function for call to 'urcl::UrDriver::UrDriver when catkin_make the robot_driver

loubron23, 2023-08-02:
[ 61%] Built target robot_state_helper
/root/detect_ws/devel/lib/liburcl_log_handler.so: undefined reference to `urcl::setLogLevel(urcl::LogLevel)'
/root/detect_ws/devel/lib/liburcl_log_handler.so: undefined reference to `urcl::registerLogHandler(std::unique_ptr<urcl::LogHandler, std::default_delete<urcl::LogHandler> >)'
/root/detect_ws/devel/lib/liburcl_log_handler.so: undefined reference to `urcl::unregisterLogHandler()'
collect2: error: ld returned 1 exit status
Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/dashboard_client.dir/build.make:220: recipe for target '/root/detect_ws/devel/lib/ur_robot_driver/dashboard_client' failed
make[2]: *** [/root/detect_ws/devel/lib/ur_robot_driver/dashboard_client] Error 1
CMakeFiles/Makefile2:991: recipe for target 'Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/dashboard_client.dir/all' failed
make[1]: *** [Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/dashboard_client.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:30:0:
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h: In constructor 'ur_driver::HardwareInterface::HardwareInterface()':
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:283:20: warning: 'ur_driver::HardwareInterface::cartesian_pose_command_' will be initialized after [-Wreorder]
   urcl::vector6d_t cartesian_pose_command_;
                    ^~~~~~~~~~~~~~~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:272:20: warning:   'urcl::vector6d_t ur_driver::HardwareInterface::joint_positions_' [-Wreorder]
   urcl::vector6d_t joint_positions_;
                    ^~~~~~~~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:58:1: warning:   when initialized here [-Wreorder]
 HardwareInterface::HardwareInterface()
 ^~~~~~~~~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function 'virtual bool ur_driver::HardwareInterface::init(ros::NodeHandle&, ros::NodeHandle&)':
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:290:100: error: no matching function for call to 'urcl::UrDriver::UrDriver(std::__cxx11::string&, std::__cxx11::string&, std::__cxx11::string&, std::__cxx11::string&, std::_Bind_helper<false, void (ur_driver::HardwareInterface::*)(bool), ur_driver::HardwareInterface*, const std::_Placeholder<1>&>::type, bool&, std::remove_reference<std::unique_ptr<urcl::ToolCommSetup>&>::type, uint32_t, uint32_t, int&, double&, bool&, std::__cxx11::string&, int&, int&)'
         servoj_lookahead_time, non_blocking_read_, reverse_ip, trajectory_port, script_command_port));
                                                                                                    ^
In file included from /root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:63:0,
                 from /root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:30:
/opt/ros/melodic/include/ur_client_library/ur/ur_driver.h:154:3: note: candidate: urcl::UrDriver::UrDriver(const string&, const string&, const string&, const string&, std::function<void(bool)>, bool, const string&, uint32_t, uint32_t, int, double, bool, const string&, uint32_t)
   UrDriver(const std::string& robot_ip, const std::string& script_file, const std::string& output_recipe_file,
   ^~~~~~~~
/opt/ros/melodic/include/ur_client_library/ur/ur_driver.h:154:3: note:   candidate expects 14 arguments, 15 provided
/opt/ros/melodic/include/ur_client_library/ur/ur_driver.h:123:3: note: candidate: urcl::UrDriver::UrDriver(const string&, const string&, const string&, const string&, std::function<void(bool)>, bool, std::unique_ptr<urcl::ToolCommSetup>, const string&, uint32_t, uint32_t, int, double, bool, const string&, uint32_t)
   UrDriver(const std::string& robot_ip, const std::string& script_file, const std::string& output_recipe_file,
   ^~~~~~~~
/opt/ros/melodic/include/ur_client_library/ur/ur_driver.h:123:3: note:   no known conversion for argument 8 from 'uint32_t {aka unsigned int}' to 'const string& {aka const std::__cxx11::basic_string<char>&}'
/opt/ros/melodic/include/ur_client_library/ur/ur_driver.h:92:3: note: candidate: urcl::UrDriver::UrDriver(const string&, const string&, const string&, const string&, std::function<void(bool)>, bool, std::unique_ptr<urcl::ToolCommSetup>, uint32_t, uint32_t, int, double, bool, const string&, uint32_t)
   UrDriver(const std::string& robot_ip, const std::string& script_file, const std::string& output_recipe_file,
   ^~~~~~~~
/opt/ros/melodic/include/ur_client_library/ur/ur_driver.h:92:3: note:   candidate expects 14 arguments, 15 provided
In file included from /root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:30:0:
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h: In constructor 'ur_driver::HardwareInterface::HardwareInterface()':
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:283:20: warning: 'ur_driver::HardwareInterface::cartesian_pose_command_' will be initialized after [-Wreorder]
   urcl::vector6d_t cartesian_pose_command_;
                    ^~~~~~~~~~~~~~~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/include/ur_robot_driver/hardware_interface.h:272:20: warning:   'urcl::vector6d_t ur_driver::HardwareInterface::joint_positions_' [-Wreorder]
   urcl::vector6d_t joint_positions_;
                    ^~~~~~~~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:58:1: warning:   when initialized here [-Wreorder]
 HardwareInterface::HardwareInterface()
 ^~~~~~~~~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function 'bool ur_driver::HardwareInterface::setIO(ur_msgs::SetIORequest&, ur_msgs::SetIOResponse&)':
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1114:31: error: 'class urcl::UrDriver' has no member named 'setToolVoltage'
     res.success = ur_driver_->setToolVoltage(static_cast<urcl::ToolVoltage>(req.state));
                               ^~~~~~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function 'bool ur_driver::HardwareInterface::zeroFTSensor(std_srvs::TriggerRequest&, std_srvs::TriggerResponse&)':
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1154:37: error: 'class urcl::UrDriver' has no member named 'zeroFTSensor'
     res.success = this->ur_driver_->zeroFTSensor();
                                     ^~~~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function 'bool ur_driver::HardwareInterface::setPayload(ur_msgs::SetPayloadRequest&, ur_msgs::SetPayloadResponse&)':
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1165:35: error: 'class urcl::UrDriver' has no member named 'setPayload'
   res.success = this->ur_driver_->setPayload(req.mass, cog);
                                   ^~~~~~~~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:1165:50: error: 'ur_msgs::SetPayloadRequest {aka struct ur_msgs::SetPayloadRequest_<std::allocator<void> >}' has no member named 'mass'
   res.success = this->ur_driver_->setPayload(req.mass, cog);
                                                  ^~~~
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp: In member function 'virtual bool ur_driver::HardwareInterface::init(ros::NodeHandle&, ros::NodeHandle&)':
/root/detect_ws/src/Universal_Robots_ROS_Driver/ur_robot_driver/src/hardware_interface.cpp:290:100: error: no matching function for call to 'urcl::UrDriver::UrDriver(std::__cxx11::string&, std::__cxx11::string&, std::__cxx11::string&, std::__cxx11::string&, std::_Bind_helper<false, void (ur_driver::HardwareInterface::*)(bool), ur_driver::HardwareInterface*, const std::_ [... cut ...] terface.cpp.o' failed
make[2]: *** [Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/ur_robot_driver_node.dir/src/hardware_interface.cpp.o] Error 1
CMakeFiles/Makefile2:1122: recipe for target 'Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/ur_robot_driver_node.dir/all' failed
make[1]: *** [Universal_Robots_ROS_Driver/ur_robot_driver/CMakeFiles/ur_robot_driver_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed

urrsk, 2023-08-07:
Hi @loubron23 
Can you please help us reproducing you issue by giving us:
- What version of ROS you are using?
- Are you building the Universal Robot Client Library from source or are you using the binary?
- That command are you calling to compile?

Currently, it looks like that your compiler can not find the client library or that the version you have installed is outdated

fmauch, 2023-08-08:
@loubron23 from your output it seems that you are building on melodic. Since melodic is end-of-life you cannot necessarily expect to run from latest master. However, since in our CI we successfully build the master branch against the released ur_client_library on melodic, I suspect it actually should work.

You could try
 - Making an update (`sudo apt update && sudo apt dist-upgrade`) to get the latest ROS packages installed
 - Removing the binary client library (`sudo apt remove ros-melodic-ur-client-library`) and building the client library also inside your workspace (Please refer to the [all-src build](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#alternative-all-source-build) for further info.
 - Switch to ROS noetic ;-) I know, there probably is a reason you are still using melodic, but in the long run I would not expect things to get any easier on ROS melodic.

closed 2023-11-30 by RobertWilbrandt