UniversalRobots/Universal_Robots_ROS_Driver #654: FIFO Scheduling notification should maybe be a Warning, not an Error for Non-RTDE Use?

vetletj, 2023-08-21:
# Summary
This issue pertains to the error messages received during the initialization of urdriver in non-RTDE use contexts. Given that FIFO scheduling might not be critical for all users, the classification of this message as an error might be misleading.

# Versions
 - ROS Noetic
 - Ubuntu 20.04

# Impact
The main impact is potential confusion for users not relying on RTDE. While the message is presented as an error, it may not be pertinent as an error for all setups and uses, particularly when RTDE isn't employed.

# Issue details
For users not leveraging RTDE, FIFO scheduling is not always crucial. The current error classification might be alarming and misleading, causing undue concern or misdirection in troubleshooting.

## Use Case and Setup
The system is being set up for non-RTDE tasks using the urdriver. Our objective is seamless initialization without encountering unnecessary error messages that are not relevant to our specific use case.

## Project status at point of discovered
In normal use during the initialization of urdriver.

## Steps to Reproduce
* Initialize urdriver without an RTDE-focused setup.
* Observe the error messages related to FIFO scheduling.

## Expected Behavior
Given that our system isn't set up for RTDE, the FIFO scheduling message should ideally appear as a warning, alerting the user but not categorizing it as a critical error.

## Actual Behavior
Upon initializing urdriver, error messages related to FIFO scheduling were presented, even though the context of use might not deem it as an error-worthy notification.

```
[ INFO] [1692625108.563732795]: Connected: Universal Robots Dashboard Server
[ INFO] [1692625108.577748350]: Initializing urdriver
[ERROR] [1692625108.578747367]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ERROR] [1692625108.578767791]: Scheduling is NOT SCHED_FIFO!
[ INFO] [1692625108.587692826]: waitForService: Service [/ur_hardware_interface/dashboard/play] is now available.
[INFO] [1692625108.751892]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1692625108.752274]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1692625109.048230086]: Negotiated RTDE protocol version to 2.
[ INFO] [1692625109.048707280]: Setting up RTDE communication with frequency 500.000000
[ INFO] [1692625110.067203653]: Checking if calibration data matches connected robot.
[ INFO] [1692625111.155662945]: Calibration checked successfully.
[ERROR] [1692625111.167347287]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
[ERROR] [1692625111.167397747]: Scheduling is NOT SCHED_FIFO!
[ INFO] [1692625111.167601269]: Loaded ur_robot_driver hardware_interface
```

## Workaround Suggestion
A potential immediate workaround is to manually filter or ignore this particular error if you are certain RTDE is not applicable in your context. However, an adjustment in the software to classify it as a warning for non-RTDE uses would be ideal.

RobertWilbrandt, 2023-08-24:
Thank you for your report - I agree that an error doesn't seem appropriate for a recommendation, changing [this](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/helpers.cpp#L49) to `URCL_LOG_WARN` seems reasonable to me. The duplicate error message there also seems wrong.

RobertWilbrandt, 2023-11-30:
With the change merged in the client library this is no longer reported as an error.

closed 2023-11-30 by RobertWilbrandt