UniversalRobots/Universal_Robots_ROS_Driver #658: joint_trajectory_controller inactive with ursim

linmeon, 2023-08-31:
Hi:
ROS2 Humble Ubuntu22.04
with doc https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/usage.html

i am testing ursim and moveit

when i run 
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=<UR_TYPE> robot_ip:=<IP_OF_THE_ROBOT> launch_rviz:=true
`
and I control the robot using VNC, the robot in RViz also moves. However, it doesn't work when I use MoveIt to plan and execute.

so  i  run
`ros2 control list_controllers`
shows 
`speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active
**joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] inactive**
forward_position_controller[position_controllers/JointGroupPositionController] inactive
io_and_status_controller[ur_controllers/GPIOController] active`
then i close all 
run 
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true
`
The control through VNC cannot synchronize with RViz, but the planning in MoveIt is consistent with RViz.
the  run
`ros2 control list_controllers`
shows
`speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active
**joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] active**
forward_position_controller[position_controllers/JointGroupPositionController] inactive
io_and_status_controller[ur_controllers/GPIOController] active`

I also tried the example 
`ros2 launch ur_robot_driver test_joint_trajectory_controller.launch.py
`
same result 

i also tried 
`Fake hardware on ROS2 Galactic[](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/usage.html#fake-hardware-on-ros2-galactic)
Currently, the scaled_joint_trajectory_controller does not work on ROS2 Galactic. There is an [upstream Merge-Request](https://github.com/ros-controls/ros2_control/pull/635) pending to fix that. Until this is merged and released, please change the default controller in the [controllers.yaml](https://docs.ros.org/en/ros2_packages/humble/api/ur_robot_driver/ur_moveit_config/config/controllers.yaml) file. Make sure that the default field is assigned true for the joint_trajectory_controller and false for the scaled_joint_trajectory_controller.`
also not work 
what should i do?

thank you!

linmeon, 2023-08-31:
UR_TYPE = ur3

fmauch, 2023-08-31:
Since this is the ROS 1 driver's repo and I replied to your [issue in the ROS 2 driver repo](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/789), I'll close this here.