UniversalRobots/Universal_Robots_ROS_Driver #673: Failed connection between pc and ur3 robot

MagicBY, 2023-11-20:
### Affected ROS Driver version(s)

Universal_ROS_driver

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

URSoftware 3.9.0.64176(Mar 01 2019)

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
Failed to connect the pc and the CB3 robot via external control urcap

## Current status
Hi, I am connecting the pc and CB3 robot using a WAN cable. The robot's IP address is set as [192.168.1.10]. And I could ping the robot from the pc, the response is as follows. 
"$ ping 192.168.1.10
PING 192.168.1.10 (192.168.1.10) 56(84) bytes of data.
64 bytes from 192.168.1.10: icmp_seq=1 ttl=64 time=0.858 ms
64 bytes from 192.168.1.10: icmp_seq=2 ttl=64 time=1.69 ms
64 bytes from 192.168.1.10: icmp_seq=3 ttl=64 time=0.983 ms
^C
--- 192.168.1.10 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2025ms
rtt min/avg/max/mdev = 0.858/1.175/1.685/0.363 ms

$ nmap -p29999-50004 192.168.1.10
Starting Nmap 7.80 ( https://nmap.org ) at 2023-11-20 14:43 GMT
Nmap scan report for 192.168.1.10
Host is up (0.000080s latency).
Not shown: 19997 closed ports
PORT      STATE SERVICE
29999/tcp open  bingbang
30001/tcp open  pago-services1
30002/tcp open  pago-services2
30003/tcp open  amicon-fpsu-ra
30004/tcp open  amicon-fpsu-s
30011/tcp open  unknown
30012/tcp open  unknown
30013/tcp open  unknown
33195/tcp open  unknown

Nmap done: 1 IP address (1 host up) scanned in 0.16 seconds" .It means the robot and the robot is connected. 

# Setup
On the robot side, it installed three URCaps file: External control; Robotiq_Copilot; Robotiq_Grippers. 

On my pc side, I am using Ubuntu 20.04 ROS Noetic. 

# Issue details
I first execute the command and here is the response. 
[... 316 lines cut ...]
[ INFO] [1700491562.365947707]: wrist_2_joint type: RotAxis
[ INFO] [1700491562.365954668]: wrist_3_joint type: RotAxis
[ INFO] [1700491562.365962001]: Number of joints in handle = 6
[ INFO] [1700491562.365969505]: kdl_chain.getNrOfJoints: 6
[ INFO] [1700491562.365976088]: Finished Kinematic Base init
[INFO] [1700491562.366206]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1700491562.368021]: Loading controller: pos_joint_traj_controller
[INFO] [1700491562.372001]: Loading controller: joint_state_controller
[INFO] [1700491562.388183]: Loading controller: joint_group_vel_controller
[INFO] [1700491562.396067]: Loading controller: force_torque_sensor_controller
[INFO] [1700491562.403948]: Controller Spawner: Loaded controllers: pos_joint_traj_controller, joint_group_vel_controller
[INFO] [1700491562.412083]: Controller Spawner: Loaded controllers: cartesian_velocity_controller, joint_state_controller, force_torque_sensor_controller
[INFO] [1700491562.419992]: Started controllers: cartesian_velocity_controller, joint_state_controller, force_torque_sensor_controller
" 

Then I start the remoteROScontrol.urcap program from the teaching pandent. Here is the error "robotiq copilot: RTDE interface cannot initialize. Make sure that Ethernet/IP adapter and profinet are disabled, and that no other URCaps use the RTDE interface." And I cannot skip this problem therefore the external control is not established.

How may I solve this problem? Your advice is greatly appreciated.

Regards,
Boyu

### Accept Public visibility

- [X] I agree to make this context public

urrsk, 2023-11-23:
Hi @MagicBY 
It seems that there is at least one conflicting RTDE input subscription between the ROS driver and the Robotiq Copilot URCap.
Only one RTDE connection can control an input register at a time.
You can see which RTDE input subscriptions the ROS driver depending on here: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/resources/rtde_input_recipe.txt

The same will be the case if you have enabled Ethernet/IP or Profinet fieldbus feature.

Hope this help

MagicBY, 2024-01-18:
Hi @urrsk,
Thanks for your reply. I couldn't find a way to diagnose this RTDE confliction both on my pc and ur3 robot teaching pendant. I reinstalled the ubuntu system and ROS. Now I still came across with the issue of connection between pc and robot. Here is a more detailed error message.

From my pc command window, it seems mostly fine. 
boyu@boyu:~/catkin_ws$ roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.1.10 kinematics_config:=/home/boyu/catkin_ws/src/UR3_resources/ur3_calibration_lab.yaml
... logging to /home/boyu/.ros/log/986a2046-b541-11ee-a103-1d9f8af5ffd7/roslaunch-boyu-20898.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
started roslaunch server http://boyu:43253/

SUMMARY
========

PARAMETERS
 * /cartesian_velocity_controller/publish_rate: 125
 * /cartesian_velocity_controller/root_name: base_link
 * /cartesian_velocity_controller/tip_name: wrist_3_link
 * /cartesian_velocity_controller/type: cartesian_control...
 * /controller_stopper/consistent_controllers: ['cartesian_veloc...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 125
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
 * /joint_based_cartesian_traj_controller/type: position_controll...
 * /joint_group_vel_controller/joints: ['shoulder_pan_jo...
 * /joint_group_vel_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 125
 * /joint_state_controller/type: joint_state_contr...
 * /pos_joint_traj_controller/action_monitor_rate: 20
 * /pos_joint_traj_controller/constraints/elbow_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/elbow_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/goal_time: 0.6
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_lift_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/shoulder_pan_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /pos_joint_traj_controller/constraints/wrist_1_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_1_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_2_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_2_joint/trajectory: 0.2
 * /pos_joint_traj_controller/constraints/wrist_3_joint/goal: 0.1
 * /pos_joint_traj_controller/constraints/wrist_3_joint/trajectory: 0.2
 * /pos_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /pos_joint_traj_controller/state_publish_rate: 125
 * /pos_joint_traj_controller/stop_trajectory_duration: 0.5
 * /pos_joint_traj_controller/type: position_controll...
 * /pose_based_cartesian_traj_controller/base: base
 * /pose_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /pose_based_cartesian_traj_controller/tip: tool0_controller
 * /pose_based_cartesian_traj_controller/type: pose_controllers/...
 * /robot_description: <?xml version="1....
 * /robot_status_controller/handle_name: industrial_robot_...
 * /robot_status_controller/publish_rate: 10
 * /robot_status_controller/type: industrial_robot_...
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /scaled_pos_joint_traj_controller/action_monitor_rate: 20
[... 250 lines cut ...]
[ INFO] [1705500386.410740150]: Getting joint handles
[ INFO] [1705500386.410756451]: shoulder_pan_joint type: RotAxis
[ INFO] [1705500386.410775793]: shoulder_lift_joint type: RotAxis
[ INFO] [1705500386.410826328]: elbow_joint type: RotAxis
[ INFO] [1705500386.410838974]: wrist_1_joint type: RotAxis
[ INFO] [1705500386.410852020]: wrist_2_joint type: RotAxis
[ INFO] [1705500386.410865316]: wrist_3_joint type: RotAxis
[ INFO] [1705500386.410880050]: Number of joints in handle = 6
[ INFO] [1705500386.410891178]: kdl_chain.getNrOfJoints: 6
[ INFO] [1705500386.410898961]: Finished Kinematic Base init
[INFO] [1705500386.413713]: Loading controller: joint_state_controller
[INFO] [1705500386.421755]: Loading controller: force_torque_sensor_controller
[INFO] [1705500386.429760]: Controller Spawner: Loaded controllers: cartesian_velocity_controller, joint_state_controller, force_torque_sensor_controller
[INFO] [1705500386.437635]: Started controllers: cartesian_velocity_controller, joint_state_controller, force_torque_sensor_controller
[ INFO] [1705500390.611482761]: Robot requested program
[ INFO] [1705500390.611539799]: Sent program to robot

However, when I run the remoteROScontrol.urcap file on the teaching pendant, after I press the play button, here is the error message. Please see the "play.jpg". If I press the fast forward key, the error message is shown in the "fast_forward_key.jpg".

I checked every files relate to the ur3_bringup.launch, but couldn't find the "tool_contact" variables. I am looking for advice on how to diagnose this problem and how to solve them. Thanks a lot.

Kind regards,
Boyu
[image]
[image]

urrsk, 2024-02-20:
@MagicBY what software version is installed on your robot.
You need to use at least SW version 3.14.3 for CB3

fmauch, 2024-07-03:
Closing due to inactivity. Please feel free to comment / reopen.

closed 2024-07-03 by fmauch