UniversalRobots/Universal_Robots_ROS_Driver #675: Problem running test_move code of ur robot driver

Elarak, 2023-12-05:
### Affected ROS Driver version(s)

Latest

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

3.14

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

I would like to be able to control an UR robot. I use a virtual box vm to have linux. I connected the robot and my computer via ethernet. The IPV4 address is: 192.168.0.140. My Vm is connected via NAT and Ethernet bridge and has the IPVA address: 192.168.0.137.
When I go to my vm, and ping 192.168.0.140, I receive the data. When I want to see the real position of the robot on my computer with rviz, I see it.
The external control has a comment parameter: IP 192.168.0.137 and port 50002.
But when I run the program on the external control's polyscope, I get the error on the terminal:

[ INFO] [1701765995.502192558]: Robot connected to reverse interface. Ready to receive control commands.
[ WARN] [1701765995.598648173]: Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.
[ERROR] [1701765995.632654285]: Sending data through socket failed.

### Accept Public visibility

- [X] I agree to make this context public

Elarak, 2023-12-07:
When I try to connect to the headless_mode, I have :
[INFO] [1701955996.347761]: Controller Spawner: Loaded controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[INFO] [1701955996.354172]: Started controllers: joint_state_controller, scaled_pos_joint_traj_controller, speed_scaling_state_controller, force_torque_sensor_controller
[ERROR] [1701955996.423552930]: Sending data through socket failed.
[ INFO] [1701955996.423777966]: Connection to reverse interface dropped.

When I run rqt, I have a white screen : rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller 
When I echo the topic scaled_pos_joint , I have nothing : rostopic echo /scaled_pos_joint_traj_controller/state 
I set up a force sensor on my UR3 and I can read the data.

urrsk, 2023-12-20:
Hi @Elarak 

> [ INFO] [1701765995.502192558]: Robot connected to reverse interface. Ready to receive control commands.
[ WARN] [1701765995.598648173]: Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.

it sounds like a re-connection problem.

You can help us reproduce it by detailed describing:
- How have you setup your VM network.
This is normally the source for connectivity problems when running the driver on a VM.
- How you start your ros nodes and specially the UR ROS driver
- Can you give us the full log on both the UR robot and UR ROS Driver

@RobertWilbrandt do you have other needs to help on this issue?

fmauch, 2024-07-03:
I'll close this due to inactivity. Please feel free to comment / reopen if the issue still persists.

closed 2024-07-03 by fmauch