UniversalRobots/Universal_Robots_ROS_Driver #682: The tool0 pose read using tf in ros is inconsistent with the pose displayed on the teaching pendant.

liujun0808, 2024-01-14:
### Affected ROS Driver version(s)

2.0.0

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

3.15

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
The tool0 pose read using tf in ros is inconsistent with the pose displayed on the teaching pendant.

## Issue details
*Detailed description help us understand the problem. Code are welcome!*
The length of the jaws I added is 14cm. Only set tool0 in the urdf file because this gripper does not have a corresponding ros driver package.Now I can control the actual UR10 robotic arm through the moveit control in rviz, but the posture of tool0 read in the test program I wrote is inconsistent with that in the teach pendant. The direction seems to be opposite. How to solve it? I've tried changing the rpy value at tool0, but the results are always wrong. Thank you for your help!
The following is the code I changed at tool0 in urdf.
**********
    
    <link name="${prefix}flange" />
    <joint name="${prefix}wrist_3-flange" type="fixed">
      <parent link="${prefix}wrist_3_link" />
      <child link="${prefix}flange" />
      <origin xyz="0 0 0" rpy="0 0 0" />
    </joint>

    
    <link name="${prefix}tool0"/>
    <joint name="${prefix}flange-tool0" type="fixed">
      <origin xyz="0 0 0.14" rpy="0 0 0" />
      <parent link="${prefix}flange"/>
      <child link="${prefix}tool0"/>
    </joint>
[image]
[image]

### Accept Public visibility

- [X] I agree to make this context public

liujun0808, 2024-03-14:
Thank you for your answer. I solved this problem a few months ago. Sorry, I forgot how to solve it, but it should be that the coordinates were modified in ROS.

liujun0808, 2024-03-14:
> The `tool0` frame will always be in the same position on the robot flange, regardless of the settings you did on the teach pendant. If you care about getting the same transformation in ROS as you see on the teach pendant, you have to do a tf lookup from `base` to `tool0_controller`. Does this work for you?
> 
> For reference: #174

Thank you for your answer. I solved this problem a few months ago. Sorry, I forgot how to solve it, but it should be that the coordinates were modified in ROS.

RobertWilbrandt, 2024-03-18:
I'll go ahead and close this issue then, if you remember how exactly you solved this problem please feel free to leave it here anyway if more people have similar issues.

closed 2024-03-18 by RobertWilbrandt