UniversalRobots/Universal_Robots_ROS_Driver #683: Compile error: name 'tool_contact' is not defined

MagicBY, 2024-01-18:
### Affected ROS Driver version(s)

Universal_Robots_ROS_driver

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux

### How is the UR ROS Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

none

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I am running remoteROScontrol.urcap file, when I press the play button on the teaching pendant, the error message shows that "Compile error: name 'tool_contact' is not defined"

## Steps to Reproduce
I connected the pc and CB3 robot using a WAN cable. I was able to ping the robot's IP address on my pc, proving that the connection works fine.

I disabled the Ethernet/IP or Profinet fieldbus feature

Then I launch the ur3_bringup.launch file on my pc, here is what shown on the command window.
boyu@boyu:~/catkin_ws$ roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.1.10 kinematics_config:=/home/boyu/catkin_ws/src/UR3_resources/ur3_calibration_lab.yaml
... logging to /home/boyu/.ros/log/986a2046-b541-11ee-a103-1d9f8af5ffd7/roslaunch-boyu-20898.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
started roslaunch server http://boyu:43253/

SUMMARY
========

PARAMETERS
 * /cartesian_velocity_controller/publish_rate: 125
 * /cartesian_velocity_controller/root_name: base_link
 * /cartesian_velocity_controller/tip_name: wrist_3_link
 * /cartesian_velocity_controller/type: cartesian_control...
 * /controller_stopper/consistent_controllers: ['cartesian_veloc...
 * /force_torque_sensor_controller/publish_rate: 125
 * /force_torque_sensor_controller/type: force_torque_sens...
 * /forward_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_cartesian_traj_controller/type: pass_through_cont...
 * /forward_joint_traj_controller/joints: ['shoulder_pan_jo...
 * /forward_joint_traj_controller/type: pass_through_cont...
 * /hardware_control_loop/loop_hz: 125
 * /joint_based_cartesian_traj_controller/base: base
 * /joint_based_cartesian_traj_controller/joints: ['shoulder_pan_jo...
 * /joint_based_cartesian_traj_controller/tip: tool0
[... 284 lines cut ...]
[ INFO] [1705500386.410726855]: Getting limits for joint: shoulder_pan_joint
[ INFO] [1705500386.410740150]: Getting joint handles
[ INFO] [1705500386.410756451]: shoulder_pan_joint type: RotAxis
[ INFO] [1705500386.410775793]: shoulder_lift_joint type: RotAxis
[ INFO] [1705500386.410826328]: elbow_joint type: RotAxis
[ INFO] [1705500386.410838974]: wrist_1_joint type: RotAxis
[ INFO] [1705500386.410852020]: wrist_2_joint type: RotAxis
[ INFO] [1705500386.410865316]: wrist_3_joint type: RotAxis
[ INFO] [1705500386.410880050]: Number of joints in handle = 6
[ INFO] [1705500386.410891178]: kdl_chain.getNrOfJoints: 6
[ INFO] [1705500386.410898961]: Finished Kinematic Base init
[INFO] [1705500386.413713]: Loading controller: joint_state_controller
[INFO] [1705500386.421755]: Loading controller: force_torque_sensor_controller
[INFO] [1705500386.429760]: Controller Spawner: Loaded controllers: cartesian_velocity_controller, joint_state_controller, force_torque_sensor_controller
[INFO] [1705500386.437635]: Started controllers: cartesian_velocity_controller, joint_state_controller, force_torque_sensor_controller
[ INFO] [1705500390.611482761]: Robot requested program
[ INFO] [1705500390.611539799]: Sent program to robot

Then I press the play button on the teaching pendant, it showed "Compile error: name 'tool_contact' is not defined".

[image]

### Accept Public visibility

- [X] I agree to make this context public

urrsk, 2024-03-11:
You are most likely using a software version for the robot that is pretty old.
Please update the robot Software to the latest version or at least one equal or later than 3.14.3, See https://github.com/UniversalRobots/Universal_Robots_Client_Library?tab=readme-ov-file#requirements

You can download the update from: https://www.universal-robots.com/download/?filters[]=98763&query=

MagicBY, 2024-05-06:
It worked after I upgraded the system. Thanks @urrsk

fmauch, 2024-05-07:
Glad to hear that!

closed 2024-05-07 by fmauch