UniversalRobots/Universal_Robots_ROS_Driver #684: A skip program line or continue option

RyanPaulMcKenna, 2024-01-24:
### Feature summary

Add continue and skip options to /ur_hardware_interface/dashboard/.
Currently we can start and stop programs like so;
 rosservice call /ur_hardware_interface/dashboard/play
 rosservice call /ur_hardware_interface/dashboard//stop

It would be good to be able to skip a single line of a program or have play from line or jump to the next line.

The reason I think this would be useful is it would allow us to use the external control urcap program and then when we want to do other things that external control can't support we can skip past or continue past that line.

RobertWilbrandt, 2024-03-14:
Have you tried the [/ur_hardware_interface/hand_back_control](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/ROS_INTERFACE.md#hand_back_control-std_srvstrigger) service? This will return from the external control block on the teach pendant. For example, in the following example i could use it to return from external control, show a popup on the screen and after it was pressed it would restart into external control:
[image]

RyanPaulMcKenna, 2024-03-22:
I was using that something similar but I have found a simpler way that is purely programmatic. Thanks for the suggestion though, should I close the issue?

RobertWilbrandt, 2024-03-27:
Glad to hear you found a solution, feel free to open another issue if you encounter further problems.

closed 2024-03-27 by RobertWilbrandt