UniversalRobots/Universal_Robots_ROS_Driver #687: Test move fails, and loud sound of UR10e real robot appears

wyifanjeff, 2024-03-13:
### Affected ROS Driver version(s)

noetic

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot, Real robot

### Robot SW / URSim version(s)

5.14

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

I followed the tutorial for the ROS UR driver (https://github.com/UniversalRobots/Universal_Robots_ROS_Driver), but did not get the calibration. However, in usage_example, when I want to control the robot using the test_move script, the code shows that everything is fine, but the robot cannot move and makes a loud sound.

Have any of you encountered this, or does anyone know what might be wrong?

### Accept Public visibility

- [X] I agree to make this context public

RobertWilbrandt, 2024-03-14:
Can you please provide us with the logs for your failed calibration extraction and your testing problem? In particular, it would be helpful to see if you can correctly connect to the robot.

Based on not being able to extract the calibration, i would suspect there to be some network problem (in particular because running in a VM adds an extra layer of complexity here), but i don't know what could cause the robot to make a sound. Be aware that we generally recommend to run the driver directly on a ubuntu PC with realtime patches applied as described [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md).

fmauch, 2024-04-17:
I'll mark this to get closed if no further input is provided.

Additionally to the things requested by @RobertWilbrandt: Could you please post the actual commands and their output that you used to start the driver and the test motion?

closed 2024-05-08 by fmauch