UniversalRobots/Universal_Robots_ROS_Driver #689: Connecting to UR3 with WSL causes compile error: "null" not defined

NathzxcCode, 2024-03-18:
### Affected ROS Driver version(s)

latest from github

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Windows using WSL2

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

3.15

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
i am trying to control the UR3 robot using WSL however i keep getting issues when running the external control program getting connection to remote PC at 192.168.1.102:50002 could not be established. Reason connection timed out. i followed the solution https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/486 adding portforwarding and turning the firewall off. however i get a new issue Compile error name "null" is not defined. this error seems to appear when i add my windows IP as the reverse_ip:=192.168.1.102 when loading the driver. 

## Issue details
i have done portforwarding:
netsh interface portproxy add v4tov4 listenport=50002 connectport=50002 connectaddress=172.17.96.1
netsh interface portproxy add v4tov4 listenport=50001 connectport=50001 connectaddress=172.17.96.1
netsh interface portproxy add v4tov4 listenport=50003 connectport=50003 connectaddress=172.17.96.1

set my external control ip and port to 192.168.1.102 50002
i run the driver: roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.1.101 reverse_ip:=192.168.1.102
i can see the robot in rvis and track how it moves. however when i try to run the extrenal control program (external control urcap in the program tree) i get the error mentioned.
[image]

when i dont pass the reverse_ip argument into the driver i get the error connection to the remote pc at ip:port could not be established. reason: connection timed out.
what do i need to do in order to control my robot through WSL?
could headless mode work better if i keep getting errors with the external control URCap?

### Accept Public visibility

- [X] I agree to make this context public

RobertWilbrandt, 2024-03-19:
Can you please verify that this is your current setup?
- Robot ip: `192.168.1.101`
- Windows static ip: `192.168.1.102`
- Firewall disabled
- Port forwarding done for ports `50001`, `50002`, `50001` with `netsh interface portproxy add v4tov4 listenport=50002 connectport=<port> connectaddress=<wsl ip>`
- External control ip set to `192.168.1.102` 
- Driver started using `roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.1.101 reverse_ip:=192.168.1.102`

I am not quite sure what could be missing in this case, i'll try to test it out. Im am not really sure how a `null` appears in the program. Using headless mode should not really change anything, as it only loads the external control script automatically. Connection problems from within the script should be unaffected by this.

NathzxcCode, 2024-03-19:
Hi Robert,
Yes this is my current setup

NathzxcCode, 2024-03-19:
One thing, I'm not sure of it helps but I have a ethernet connection tonight from the robot directly to my laptop through a USB ethernet adapter, no router.
I have setup the iPhone so that it is static to this address tho and can ping the robot still

RobertWilbrandt, 2024-03-21:
To me this looks like there is some problem transmitting the urscript from `external_control.urscript` to the urcap. The urcap then doesn't handle this issue correctly and outputs `null` into the program before executing.

I am a little bit confused about the port assignment here - in the non-headless mode, i would expect traffic to go through port `50002`, in headless mode we directly send the program through the secondary interface at port `30002`. Even so, could you please check if you need to additionally forward ports `30001`, `30002` and `30004` (see [driver architecture](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/master/ur_robot_driver#technical-details) and [client interface overview](https://www.universal-robots.com/articles/ur/interface-communication/overview-of-client-interfaces/))? @urrsk do you maybe have some more insight on this?

NathzxcCode, 2024-03-21:
Hi thanks, 
I will try today to see if forwarding ports 30001, 2, 4 makes any difference. 
im just curious in the ur3_bringup.launch there is also a port 50004 <arg name="script_command_port" default="50004" should pass this though too? 
also on the client interface overview link you sent it mentions that ports 50003,4,5 are only used on e-series robots does this effect things if im using a ur3
[image]

also just to clarify when you say to disable the firewall you are referring to windows firewall correct? WSL also seems to have a firewall however incoming traffic on that is set to allow.
thanks

NathzxcCode, 2024-03-21:
I tested passing through ports 30001,2,4 with no luck.
i disabled the WSL firewall aswell as windows firewall with no luck
i tried headless_mode = True, it ran and seemed to start the external control script but i was unable to execute the rosrun ur_robot_driver test_move

troyCor, 2024-06-19:
Hello,

Has there been any update on this issue? I am running into the same issue using a CB3 UR10. After following the debugging steps in #486 I have ended up here with the same issue as NathzxcCode

NathzxcCode, 2024-06-19:
I ended up dualbooting my system with Linux because I was against a tight deadline. I didn't end up trying the latest suggestions

urrsk, 2024-06-21:
> Hello,
> 
> Has there been any update on this issue? I am running into the same issue using a CB3 UR10. After following the debugging steps in #486 I have ended up here with the same issue as NathzxcCode
> 
> I can ping the robot on 192.168.1.102 and its pose updates live in RVIZ
> 
> out put from `telnet ` is `w$w$�?�333333@9?ə�����?ə�����?ə�����?ə�����?ə� ...

@troyCor okay it seems that you have the right ports open and can connect to the robot.
The External control urcap uses by default port 50002 to receive the urscript from the driver.
Please check if the ROS driver is running and also check if there is some firewall or similar that blokes the communication for the ROS driver.

Please share when you find the solution

urrsk, 2024-06-21:
> I ended up dualbooting my system with Linux because I was against a tight deadline. I didn't end up trying the latest suggestions

@NathzxcCode happy you got forward and a dual boot was an option for you. This is also the solution the in regards real time performance as well.
Though, maybe a docker container could also be a similar solution getting it running on windows

urfeex, 2026-05-04:
Closing this due to inactivity. Feel free to comment if there has been something new coming up since last posted.