UniversalRobots/Universal_Robots_ROS_Driver #691: Control Robotiq 2F-85 gripper on UR5e

Yeesha-R, 2024-04-01:
### Affected ROS Driver version(s)

ROS noetic

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux with realtime patch

### How is the UR ROS Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

5.12.4

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I am unable to control the Robotiq 2F-85 gripper directly connected to the tool flange of my UR5e robotic arm.

## Issue details
I have a setup in which the 2f-85 gripper is directly attached to the tool flange of my UR5e robot arm. I am relatively new to ROS, but I have read most issues on this topic, including #240 but I still get errors.

## Steps to Reproduce

To start the robotic arm, I run
`roslaunch ur_robot_driver ur5e_bringup.launch robot_ip:=192.168.1.2 kinematics_config:=$(rospack find ur_calibration)/my_robot_calibration.yaml use_tool_communication:=true tool_device_name:=/tmp/ttyUR tool_voltage:=24`

This creates a file named 'ttyUR' in my tmp folder.

However, when I run `rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py /tmp/ttyUR`, I encounter the following error and notice that the ttyUR file created in my tmp folder gets deleted

```
Traceback (most recent call last):
File "/home/ubuntu/catkin_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 89, in
mainLoop(sys.argv[1])
File "/home/ubuntu/catkin_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 75, in mainLoop
status = gripper.getStatus()
File "/home/ubuntu/catkin_ws/src/robotiq/robotiq_2f_gripper_control/src/robotiq_2f_gripper_control/baseRobotiq2FGripper.py", line 107, in getStatus
status = self.client.getStatus(6);
File "/home/ubuntu/catkin_ws/src/robotiq/robotiq_modbus_rtu/src/robotiq_modbus_rtu/comModbusRtu.py", line 94, in getStatus
response = self.client.read_holding_registers(0x07D0, numRegs, unit=0x0009)
File "/home/ubuntu/.local/lib/python3.8/site-packages/pymodbus/client/mixin.py", line 107, in read_holding_registers
return self.execute(
File "/home/ubuntu/.local/lib/python3.8/site-packages/pymodbus/client/base.py", line 391, in execute
raise ConnectionException(f"Failed to connect[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusSerialClient /tmp/ttyUR:0]

```

I have also modified my launch file as follows:

```
<?xml version="1.0"?>
<launch>
  <arg name="debug" default="false" doc="Debug flag that will get passed on to ur_common.launch"/>
  <arg name="robot_ip" default="192.168.1.2" doc="IP address by which the robot can be reached."/>
  <arg name="reverse_ip" default="" doc="IP of the driver, if set to empty it will detect it automatically."/>
  <arg name="reverse_port" default="50001" doc="Port that will be opened by the driver to allow direct communication between the driver and the robot controller."/>
  <arg name="script_sender_port" default="50002" doc="The driver will offer an interface to receive the program's URScript on this port. If the robot cannot connect to this port, `External Control` will stop immediately."/>
  <arg name="trajectory_port" default="50003" doc="Port that will be opened by the driver to allow trajectory forwarding."/>
  <arg name="script_command_port" default="50004" doc="Port that will be opened by the driver to allow forwarding script commands to the robot."/>
  <arg name="tf_prefix" default="" doc="tf_prefix used for the robot."/>
  <arg name="controllers" default="pose_based_cartesian_traj_controller scaled_pos_joint_traj_controller joint_state_controller speed_scaling_state_controller force_torque_sensor_controller" doc="Controllers that are activated by default."/>
  <arg name="stopped_controllers" default="joint_based_cartesian_traj_controller forward_cartesian_traj_controller pos_joint_traj_controller joint_group_vel_controller" doc="Controllers that are initally loaded, but not started."/>
  <arg name="controller_config_file" default="$(find ur_robot_driver)/config/ur5e_controllers.yaml" doc="Config file used for defining the ROS-Control controllers."/>
  <arg name="robot_description_file" default="$(find ur_description)/launch/load_ur5e.launch" doc="Robot description launch file."/>
  <arg name="kinematics_config" default="$(find ur_description)/config/ur5e/my_robot_calibration.yaml" doc="Kinematics config file used for calibration correction. This will be used to verify the robot's calibration is matching the robot_description."/>
  <arg name="use_tool_communication" default="true" doc="On e-Series robots tool communication can be enabled with this argument"/>
  <arg name="tool_voltage" default="24" doc="Tool voltage set at the beginning of the UR program. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_parity" default="0" doc="Parity configuration used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_baud_rate" default="115200" doc="Baud rate used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_stop_bits" default="1" doc="Number of stop bits used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_rx_idle_chars" default="1.5" doc="Number of idle chars in RX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_tx_idle_chars" default="3.5" doc="Number of idle chars in TX channel used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_device_name" default="/tmp/ttyUR" doc="Local device name used for tool communication. Only used, when `use_tool_communication` is set to true."/>
  <arg name="tool_tcp_port" default="54321" doc="Port on which the robot controller publishes the tool comm interface. Only used, when `use_tool_communication` is set to true."/>
  <arg name="headless_mode" default="true" doc="Automatically send URScript to robot to execute. On e-Series this does require the robot to be in 'remote-control' mode. With this, the URCap is not needed on the robot."/>
  <arg name="ur_hardware_interface_node_required" default="true" doc="Shut down ros environment if ur_hardware_interface-node dies."/>

  <include file="$(find ur_robot_driver)/launch/ur_common.launch" pass_all_args="true" />
</launch>
```

## Expected Behavior
To establish a connection to the gripper without error

## Actual Behavior
Unable to establish a connection with the gripper.

Kindly let me know if you need any more information. Thank you.

### Accept Public visibility

- [X] I agree to make this context public

RobertWilbrandt, 2024-04-02:
The order you describe in "Issue details" seems wrong to me. Passing `use_tool_communication:=true` to `ur5e_bringup.launch` should already start `socat` in the `tool_communication` script (see [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/launch/ur_control.launch#L66)), so your explicit call to it seems superfluous. Regardless, you should only start the gripper driver after you started `socat`, but this would lead to a different error.

Did you set up the tool I/O on the teach pendant correctly? We currently have a setup with a ur5e and a robotiq 2f-140 connected to its tool connector and under installation->Tool I/O we configured the following settings:
- Controlled by: User
- Analog Inputs - Communication Interface
  - Communication Interface (Baud rate 115200, Parity None, Stop Bits One, RX Idle Chars 1.5, TX Idle Chars 3.5)
- Digital Output Mode
  - Digital Output Mode - Tool Output Voltage: 0
  - Dual Pin Power

Did you make sure no other URCap is installed that could interfere with the driver? In particular, you need to remove the robotiq URCap.

Yeesha-R, 2024-04-02:
@RobertWilbrandt, thank you for your response. I followed your instructions and adjusted the settings on my teach pendant, successfully launching the robot and executing rosrun ur_robot_driver tool_communication. However, I encountered an error when attempting to run `rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py `:

```
Traceback (most recent call last):
  File "/home/ubuntu/catkin_ws/src/robotiq/robotiq_2f_gripper_control/nodes/Robotiq2FGripperRtuNode.py", line 92, in <module>
    mainLoop(sys.argv[1])
IndexError: list index out of range

```
Despite my attempts to debug the code, I have been unable to resolve the issue. I apologize for any repetition, as I am not well-versed in this area. 
Could you let me know if I am supposed to run `rosrun robotiq_2f_gripper_control Robotiq2FGripperRtuNode.py `? If not, could you please provide guidance on how to communicate with the robot once the connection is established?

Thank you so much for your assistance.

RobertWilbrandt, 2024-04-08:
Glad to hear that tool forwarding now seems to work for you. I am not aware of the error you are seeing now, looking back at the last time we used this gripper in ROS 1 (our current setup uses ROS 2) that was the node we started.

I think you need to pass the serial device (i.e. `/tmp/ttyUR`) to the `rosrun` command as you described above to get through this error.

Yeesha-R, 2024-04-11:
Okay, thanks @RobertWilbrandt

fmauch, 2024-04-17:
@Yeesha-R Does hat mean the issue is resolved for you and can be closed?