UniversalRobots/Universal_Robots_ROS_Driver #692: Cannot connect to robot

trngqung14, 2024-04-02:
Failed to connect to robot on IP 192.168.0.101. Please check that the robot is booted and reachable on 192.168.0.101.

I have done ping, source, and check IP address and network in Polyscope and it's still error

RobertWilbrandt, 2024-04-02:
Please add some more information to the issue:
- What is your host IP address? Are you running the driver directly on a Ubuntu 20.04? Does `ping 192.168.0.101` work?
- Please provide the driver log output

trngqung14, 2024-04-02:
my host IP in Polyscope is 192.168.0.101, Polyscope version is 3.15.5,
[image]
[image]
[image]
 i am using ubuntu 20.04 on VM and ping worked
i have set ethernet in window 192.168.0.101, changed network of VM to Bridge Adapter and set IP in linux network 192.168.0.101
driver log output is below
[ERROR] [1712046723.448455422]: Failed to connect to robot on IP 192.168.0.101. Please check that the robot is booted and reachable on 192.168.0.101. Retrying in 10 seconds

Yeesha-R, 2024-04-02:
It seems that your robot's IP address is the same as your computer's IP address. This could be the potential cause of the error.

trngqung14, 2024-04-02:
@Yeesha-R when i change IP of computer and network in linux to 192.168.0.10 like the hostname in Polyscope, it still does not work. 
my robot is ur5 cb series but output remains about cb3 and e series.
[image]

Yeesha-R, 2024-04-02:
@trngqung14, I do not have a solution for this error. However, you can take a look at issue [#510](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/510) to see if it provides any assistance.

trngqung14, 2024-04-04:
Im still cannot fix this problem

Yeesha-R, 2024-04-05:
@trngqung14 according to the error prompt, the issue could be due to the software version of your polyscope. Could you check if the software version of your polyscope is at least 3.12.0?

RobertWilbrandt, 2024-04-08:
I still don't get what your network setup looks like right now. What exactly are the ip addresses of:
- The robot (probably `192.168.0.101`)?
- Your host?
- The ubuntu VM?
- Is the host ip for the external control program correctly set to the ip of your ubuntu vm?

All of these need to be different, your network interface in the vm needs to be set to Bridge Adapter and you have to be able to ping the robot from your ubuntu VM.

You say that you currently have polyscope version 3.15.5 installed, which should be fine (see [requirements](https://github.com/UniversalRobots/Universal_Robots_Client_Library?tab=readme-ov-file#requirements)). Do you have a different program or PLC trying to control the UR IOs? This is currently not supported.

closed 2024-04-17 by fmauch