UniversalRobots/Universal_Robots_ROS_Driver #693: UR10e robot vibrates and makes noise during teleoperation via Twist Controller

shrutichakraborty, 2024-04-04:
Hello all, 
I am using a UR10e robot with ros1 noetic, I have installed the drivers for the robot and I am teleoperating the robot by sending `twist` commands through the `ros_controllers_cartesian/TwistController`. However, the robot vibrates a lot and makes noise. I have tried the same with a UR10 robot to verify the source of this issue and the UR10 robot does not make any noise or vibrate, rather it smoothly follows the teleop. Could anyone point to what the issue could be? 

One of the sources could be an issue in the controller parameters (gain values of the PID), but on the `ur10e_controllers.yaml` file there are no PID parameters specified. If anyone could provide a background on how the twist controller works that would be great. For instance does it have an integrator within to send position commands to the robot?  

Thanks!

RobertWilbrandt, 2024-04-08:
I'll try reproducing this on our UR10e within the next days.

You can find informations on the twist controller [here](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian). The controller basically only forwards the twist to the hardware interface, which sends the command to the hardware. All control is performed on the robot itself, so i find it surprising that there are instabilities. The only parameter is defined [here](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/blob/main/twist_controller/cfg/TwistController.cfg) and should be tunable through [dynamic reconfigure](http://wiki.ros.org/dynamic_reconfigure).

shrutichakraborty, 2024-04-09:
> I'll try reproducing this on our UR10e within the next days.
>
> You can find informations on the twist controller [here](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian). [...]

Hi @RobertWilbrandt, thanks a lot! It would be great if you could test the same.  that would be great. I have noticed on my setup on bth ur10 and ur10e, only default joint values are specified and no twist gain is explicitly mentioned. The twist gain from the link seems to be a multiplicative constant. What is the purpose of multiplying the input twist with this gain and what effect does it have physically on the system? 

I have also tested a second ur10e robot and it too makes noise and vibrates, so it seems to me to be an issue on the twist controller of ur10e robots? Has anyone else experienced the same?

Hytac, 2024-04-09:
Same issue here with ur10e and ros2 humble.

EPVelasco, 2025-02-05:
Same issue with the robotic arm UR5e with ros noetic

EPVelasco, 2025-02-05:
I have tested it with another UR5e that we have in our lab. This one has URSoftware version 5.11.5.1010327. 
The robotic arm that has the TwistController problem has URSoftware version 5.6.0.90886.

Do you think it could be the firmware version?

EPVelasco, 2025-02-06:
I have updated the firmware of the UR5e robotic arm [firmware](https://www.universal-robots.com/download/?query=software%20update&type[]=42713).
I have updated to version 5.11, and it has worked fine. Although the manufacturer recommends to put the latest (now is 5.21).
I also had to install again the [URMagic_VNC_Installation_V1](https://forum.universal-robots.com/t/remote-desktop-to-the-controller-ui/3826) because the remote desktop with VNC has been deleted.

urfeex, 2025-02-07:
Thank you for testing that.