UniversalRobots/Universal_Robots_ROS_Driver #700: "/ur_hardware_interface/script_command" stops program on controller

kimanton, 2024-05-15:
### Affected ROS Driver version(s)

Latest

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux without realtime patch

### How is the UR ROS Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

UR E-series robot, Real robot

### Robot SW / URSim version(s)

SW 5.16.0

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
This issue is similar to many others, but they are old so I decided to open a new one
I have UR10e with Onrobot RG6 gripper. I want to control its width and force. 
The only solution seems to be `/ur_hardware_interface/script_command`. However, when I publish `"rg_grip(100, 100, 0, False, False)"` nothing happens. When I try to send a test hello like was suggested [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/77#issuecomment-573748512). 
I made sure I am in Remote Control mode, and I updated robot software and urcaps to latest available versions just in case.  

## Issue details
I need to be able to control the gripper width and force. And I also need to robot program always running because I need External Control to control joint velocities.
As was said before, I want to control gripper through `/ur_hardware_interface/script_command`. In my case the command I want to send looks like this `rg_grip(width, force, 0, False, False)`. I use simple Python publisher script to publish the message. When I send the command, there is no reaction at all. [pkowalsk1](https://github.com/pkowalsk1) mentions [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/611#issuecomment-1422412111) in EDIT section that they also couldn't send anything when using External Control

I then followed sequence presented in #131 [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/131#issuecomment-722977898), but still there is no reaction. Logs also don't have any new messages

Then I tried to see if anything will work at all. I created a program variable `var1=10` and an installation variable `var2=10`, and tried to change assign them a new value like described in #536 [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/536#issuecomment-1139752183) and [here,p5](https://s3-eu-west-1.amazonaws.com/ur-support-site/69544/scriptManual.pdf). 
The message goes through, but the variables are not updates. Instead, the program on controller stops and I have to restart the sequence from #131.

Then I tried to send a hello popup as was suggested in #77. The popup appears on TP screen, but it also stops program and I have to go thought the sequence from #131

Finally, I updated robot software and URCaps for external control and for gripper to the latest available versions and tried all previous step again with no success

## Steps to Reproduce
1. Make sure physical connection is not established between robot and host computer (ethernet unplugged)
2. Enable Remote control mode from TP.
3. Establish physical connection between robot and host computer (ethernet plugged in)
4. Run UR_ROS_Driver.
5. Run program on controller `rosservice call /ur_hardware_interface/dashboard/play`
6. Run script to send message
```
import rospy
from std_msgs.msg import String
def main():
    node_name = "rg6_node"
    rospy.init_node(node_name, anonymous=True)
    rg6_pub = rospy.Publisher('/ur_hardware_interface/script_command', String, queue_size=1, latch=True)
    cmd = String()
    rate = rospy.Rate(1)
    while not rospy.is_shutdown():        
        # width, force, tool_index, blocking, depth_comp
        data = "rg_grip(100, 100, 0, False, False)"
        # data = "global var2 = 50"
        # data = "var1 = 50"
        # data = "popup(\"hello\")"
        cmd.data = data
        rg6_pub.publish(cmd)
        rate.sleep()
        print(data)
if __name__ == "__main__":
    main()
```

## Expected Behavior
Gripper moves, or variable changes value, or "hello" popups, and TP will show that the program is still running

## Actual Behavior
Gripper never moves, variables never change values, "hello" popups. However, when sending new variables values or sending "hello" TP shows that program stopped.

# Workaround Suggestion
I use digital outputs to trigger events which call `rg_grip` in robot program. But there are only eight digital outputs, so there is only eight combinations of (force,width) I can use. This is not feasible as a longterm solution

[pkowalsk1](https://github.com/pkowalsk1) presents [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/611#issuecomment-1422412111) another possible solution by using analog outputs

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-05-22:
I am not sure whether it is possible to do what you are trying to achieve.

As I don't have such a gripper, I could not test things properly, but here's what I did:

I modified our `[external_control.urscript](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/resources/external_control.urscript)` file locally to add a function `foo_fun` and call it once in the script:

```python
def foo_fun():                                                                                                                                                                                                                                                                                                        
┊ textmsg("foo_fun called")                                                                                                                                                                                                                                                                                           
end                                                                                                                                                                                                                                                                                                                   
                                                                                                                                                                                                                                                                                                                       
foo_fun()
```

Starting the external_control program with that prints "foo_fun called" into the logs.

However, running `rostopic pub --once /ur_hardware_interface/script_command std_msgs/String "data: 'foo_fun()'"` didn't work for me either. Since I tested with a URSim docker container, I could attach to the logs easily using `docker logs -f ursim` where it states that

```
2024-05-22_11:23:50:36907 00:07:30:510 ERROR   - RuntimeException: Compile error: name 'foo_fun' is not defined
2024-05-22_11:23:50:36909 00:07:30:510 ERROR   - TCPReceiver::parseLine: Parse Error: foo_fun()
```

I am not that familiar with this, but I assume that it is only possible to call builtin functions via that interface? In the end the OnRobot URCap will add functions (such as `rg_grip`) to the running program's preamble, so they will only be defined inside the running program. I might be wrong here, though.

kimanton, 2024-05-22:
> In the end the OnRobot URCap will add functions (such as rg_grip) to the running program's preamble, so they will only be defined inside the running program.
I think you are right

Anyway, I will continue to use `digital_out` pins for now, as it seems this problems doesn't have a proper solution

I will close this issue as complete

closed 2024-05-22 by kimanton