UniversalRobots/Universal_Robots_ROS_Driver #701: ROS Noetic "ur_calibration" catkin_make error

2000222, 2024-05-26:
### Affected ROS Driver version(s)

Ubuntu 20.04

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux

### How is the UR ROS Driver installed.

From binary packets

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

none

### How is the ROS driver used.

Others

### Issue details

# Summary
"ur_calibration" passage shows errors within the catkin_make compiling process.

## Issue details
```[
100%] Linking CXX executable /home/sarah/ur5_ws/devel/lib/ur_calibration/calibration_correction
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration.cpp.o: in function `YAML::Node::Scalar[abi:cxx11]() const':
calibration.cpp:(.text._ZNK4YAML4Node6ScalarB5cxx11Ev[_ZNK4YAML4Node6ScalarB5cxx11Ev]+0x7a): undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration.cpp.o: in function `YAML::detail::node& YAML::detail::node_data::get<char [2]>(char const (&) [2], std::shared_ptr<YAML::detail::memory_holder>)':
calibration.cpp:(.text._ZN4YAML6detail9node_data3getIA2_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE[_ZN4YAML6detail9node_data3getIA2_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE]+0xd3): undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)'
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration.cpp.o: in function `YAML::detail::node& YAML::detail::node_data::get<char [5]>(char const (&) [5], std::shared_ptr<YAML::detail::memory_holder>)':
calibration.cpp:(.text._ZN4YAML6detail9node_data3getIA5_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE[_ZN4YAML6detail9node_data3getIA5_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE]+0xd3): undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)'
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration.cpp.o: in function `YAML::detail::node& YAML::detail::node_data::get<char [6]>(char const (&) [6], std::shared_ptr<YAML::detail::memory_holder>)':
calibration.cpp:(.text._ZN4YAML6detail9node_data3getIA6_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE[_ZN4YAML6detail9node_data3getIA6_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE]+0xd3): undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)'
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration.cpp.o: in function `YAML::detail::node& YAML::detail::node_data::get<char [4]>(char const (&) [4], std::shared_ptr<YAML::detail::memory_holder>)':
calibration.cpp:(.text._ZN4YAML6detail9node_data3getIA4_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE[_ZN4YAML6detail9node_data3getIA4_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE]+0xd3): undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)'
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration.cpp.o: in function `YAML::detail::node& YAML::detail::node_data::get<char [11]>(char const (&) [11], std::shared_ptr<YAML::detail::memory_holder>)':
calibration.cpp:(.text._ZN4YAML6detail9node_data3getIA11_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE[_ZN4YAML6detail9node_data3getIA11_cEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE]+0xd3): undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)'
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration.cpp.o:calibration.cpp:(.text._ZN4YAML6detail9node_data3getINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE[_ZN4YAML6detail9node_data3getINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEERNS0_4nodeERKT_St10shared_ptrINS0_13memory_holderEE]+0xd3): more undefined references to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)' follow
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o: in function `boost::filesystem::path::parent_path() const':
calibration_correction.cpp:(.text._ZNK5boost10filesystem4path11parent_pathEv[_ZNK5boost10filesystem4path11parent_pathEv]+0x2c): undefined reference to `boost::filesystem::detail::path_algorithms::find_parent_path_size(boost::filesystem::path const&)'
/usr/bin/ld: CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o: in function `boost::filesystem::absolute(boost::filesystem::path const&, boost::filesystem::path const&)':
calibration_correction.cpp:(.text._ZN5boost10filesystem8absoluteERKNS0_4pathES3_[_ZN5boost10filesystem8absoluteERKNS0_4pathES3_]+0x3c): undefined reference to `boost::filesystem::detail::absolute(boost::filesystem::path const&, boost::filesystem::path const&, boost::system::error_code*)'
collect2: error: ld returned 1 exit status
make[2]: *** [Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/build.make:167: /home/sarah/ur5_ws/devel/lib/ur_calibration/calibration_correction] Error 1
make[1]: *** [CMakeFiles/Makefile2:13597: Universal_Robots_ROS_Driver/ur_calibration/CMakeFiles/calibration_correction.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-05-27:
Hi, you selected "From binary packets" but post a compile error? This doesn't seem to match.

Unless you actually want to contribute to this driver we highly recommend using the binary package installation as explained [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver?tab=readme-ov-file#install-from-binary-packages).

That being said it looks like there are either some incompatible library versions in your path and your compiler finds header files from one, but the linker finds symbols from the other...
Or it could be that another package in your workspace sets cmake variables that break the build process. You could try building packages in isolation e.g. by using [`catkin build`](https://catkin-tools.readthedocs.io/en/latest/verbs/catkin_build.html) instead of `catkin_make`

2000222, 2024-05-28:
Actually I think it's the problem about "ur_calibration" passage. After I utilize "catkin build" to compile, I also get the issue as follows: 
```
Errors     << ur_calibration:make /home/sarah/ur5_ws/logs/ur_calibration/build.make.000.log
In file included from /home/sarah/anaconda3/include/boost/smart_ptr/detail/requires_cxx11.hpp:9,
                 from /home/sarah/anaconda3/include/boost/smart_ptr/shared_ptr.hpp:17,
                 from /home/sarah/anaconda3/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/noetic/include/ros/forwards.h:37,
                 from /opt/ros/noetic/include/ros/common.h:37,
                 from /opt/ros/noetic/include/ros/ros.h:43,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration.h:32,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration.cpp:29:
/home/sarah/anaconda3/include/boost/bind.hpp:36:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
In file included from /home/sarah/anaconda3/include/boost/smart_ptr/detail/requires_cxx11.hpp:9,
                 from /home/sarah/anaconda3/include/boost/smart_ptr/shared_ptr.hpp:17,
                 from /home/sarah/anaconda3/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/noetic/include/ros/forwards.h:37,
                 from /opt/ros/noetic/include/ros/common.h:37,
                 from /opt/ros/noetic/include/ros/ros.h:43,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration.h:32,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration_consumer.h:35,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_consumer.cpp:29:
/home/sarah/anaconda3/include/boost/bind.hpp:36:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
In file included from /home/sarah/anaconda3/include/boost/smart_ptr/detail/requires_cxx11.hpp:9,
                 from /home/sarah/anaconda3/include/boost/smart_ptr/shared_ptr.hpp:17,
                 from /home/sarah/anaconda3/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/noetic/include/ros/forwards.h:37,
                 from /opt/ros/noetic/include/ros/common.h:37,
                 from /opt/ros/noetic/include/ros/ros.h:43,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration.h:32,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/include/ur_calibration/calibration_consumer.h:35,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:29:
/home/sarah/anaconda3/include/boost/bind.hpp:36:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp: In member function ‘bool CalibrationCorrection::writeCalibrationData()’:
/home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:111:54: warning: ‘boost::filesystem::path boost::filesystem::complete(const boost::filesystem::path&)’ is deprecated: Use absolute() instead [-Wdeprecated-declarations]
  111 |     fs::path out_path = fs::complete(output_filename_);
      |                                                      ^
In file included from /home/sarah/anaconda3/include/boost/filesystem.hpp:19,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:44:
/home/sarah/anaconda3/include/boost/filesystem/operations.hpp:353:13: note: declared here
  353 | inline path complete(path const& p)
      |             ^~~~~~~~
/home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:111:54: warning: ‘boost::filesystem::path boost::filesystem::complete(const boost::filesystem::path&)’ is deprecated: Use absolute() instead [-Wdeprecated-declarations]
  111 |     fs::path out_path = fs::complete(output_filename_);
      |                                                      ^
In file included from /home/sarah/anaconda3/include/boost/filesystem.hpp:19,
                 from /home/sarah/ur5_ws/src/Universal_Robots_ROS_Driver/ur_calibration/src/calibration_correction.cpp:44:
/home/sarah/anaconda3/include/boost/filesystem/operations.hpp:353:13: note: declared here
  353 | inline path complete(path const& p)
      |             ^~~~~~~~
[... cut ...]
[build]   Warnings:  5 packages succeeded with warnings.                               
[build]   Abandoned: None.                                                             
[build]   Failed:    1 packages failed.                                                
[build] Runtime: 17 minutes and 25.8 seconds total.                                    
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```
So I don't know how to solve the sudden problem .......

fmauch, 2024-05-28:
So we do know it comes not from other packages polluting your build. However, this seems to be a problem running particular on your machine. Our CI builds (and my local ones, as well) seem to be fine.

As I said above, another source of this could be two versions of yamlcpp installed on your system.

Going to your `catkin_ws/build/ur_calibration` folder and running `grep -r yaml-cpp **/*.txt **/*.cmake` might give you some insights.

For me it looks like this:

```
CMakeCache.txt://The directory containing a CMake configuration file for yaml-cpp.
CMakeCache.txt:yaml-cpp_DIR:PATH=/usr/lib/x86_64-linux-gnu/cmake/yaml-cpp
CMakeFiles/calibration_correction.dir/link.txt:/usr/bin/c++    -rdynamic CMakeFiles/calibration_correction.dir/src/calibration.cpp.o CMakeFiles/calibration_correction.dir/src/calibration_consumer.cpp.o CMakeFiles/calibration_correction.dir/src/calibration_correction.cpp.o  -o /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/ur_calibration/lib/ur_calibration/calibration_correction  -Wl,-rpath,/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/ur_robot_driver/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/lib:/opt/ros/noetic/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/pass_through_controllers/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/scaled_joint_trajectory_controller/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/speed_scaling_state_controller/lib: /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/ur_robot_driver/lib/libur_robot_driver_plugin.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/ur_robot_driver/lib/liburcl_log_handler.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/lib/liburcl.so /opt/ros/noetic/lib/libkdl_parser.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/pass_through_controllers/lib/libpass_through_controllers.so /opt/ros/noetic/lib/libcontroller_manager.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/scaled_joint_trajectory_controller/lib/libscaled_joint_trajectory_controller.so /opt/ros/noetic/lib/libjoint_trajectory_controller.so /opt/ros/noetic/lib/libcontrol_toolbox.so /opt/ros/noetic/lib/libdynamic_reconfigure_config_init_mutex.so /opt/ros/noetic/lib/liburdf.so -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -ltinyxml /opt/ros/noetic/lib/librosconsole_bridge.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/speed_scaling_state_controller/lib/libspeed_scaling_state_controller.so /opt/ros/noetic/lib/libclass_loader.so -lPocoFoundation -ldl /opt/ros/noetic/lib/libroslib.so /opt/ros/noetic/lib/librospack.so -lpython3.8 /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.71.0 -ltinyxml2 /opt/ros/noetic/lib/librealtime_tools.so /opt/ros/noetic/lib/libtf.so -lorocos-kdl -lorocos-kdl /opt/ros/noetic/lib/libtf2_ros.so /opt/ros/noetic/lib/libactionlib.so /opt/ros/noetic/lib/libmessage_filters.so /opt/ros/noetic/lib/libroscpp.so -lpthread /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.71.0 /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 /opt/ros/noetic/lib/librosconsole.so /opt/ros/noetic/lib/librosconsole_log4cxx.so /opt/ros/noetic/lib/librosconsole_backend_interface.so -llog4cxx /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.71.0 /opt/ros/noetic/lib/libxmlrpcpp.so /opt/ros/noetic/lib/libtf2.so /opt/ros/noetic/lib/libroscpp_serialization.so /opt/ros/noetic/lib/librostime.so /usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.71.0 /opt/ros/noetic/lib/libcpp_common.so /usr/lib/x86_64-linux-gnu/libboost_system.so.1.71.0 /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0 /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 /usr/lib/x86_64-linux-gnu/libyaml-cpp.so.0.6.2 /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/lib/liburcl.so -lpthread 
CMakeFiles/calibration_test.dir/link.txt:/usr/bin/c++    -rdynamic CMakeFiles/calibration_test.dir/test/calibration_test.cpp.o CMakeFiles/calibration_test.dir/src/calibration.cpp.o  -o /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/ur_calibration/lib/ur_calibration/calibration_test   -L/home/ubuntu/checkout/ur_noetic/catkin_ws/build/ur_calibration/gtest  -Wl,-rpath,/home/ubuntu/checkout/ur_noetic/catkin_ws/build/ur_calibration/gtest:/home/ubuntu/checkout/ur_noetic/catkin_ws/build/ur_calibration/gtest/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/ur_robot_driver/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/lib:/opt/ros/noetic/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/pass_through_controllers/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/scaled_joint_trajectory_controller/lib:/home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/speed_scaling_state_controller/lib gtest/lib/libgtest.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/ur_robot_driver/lib/libur_robot_driver_plugin.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/ur_robot_driver/lib/liburcl_log_handler.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/lib/liburcl.so /opt/ros/noetic/lib/libkdl_parser.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/pass_through_controllers/lib/libpass_through_controllers.so /opt/ros/noetic/lib/libcontroller_manager.so /home/ubuntu/checkout/ur_noetic/catkin_ws/devel/.private/scaled_joint_trajectory_controller/lib/libscaled_joint_trajectory_controller.so /opt/ros/noetic/lib/libjoint_trajectory_controller.so /opt/ros/noetic/lib/libcontrol_toolbox.so /opt/ros/noetic/lib/libdynamic_reconfigure_conf [... cut ...]
catkin_generated/package.cmake:set(ur_calibration_RUN_DEPENDS "eigen" "roscpp" "ur_client_library" "yaml-cpp" "ur_robot_driver")
```

which shows that I have `yaml-cpp 0.6.2` installed in `/usr/lib/x86_64-linux-gnu` and use the includes from `/usr/lib/x86_64-linux-gnu/cmake/yaml-cpp/../../../../../include` which apparently doesn't exist, but yaml-cpp includes are in `/usr/include/yaml-cpp/` so it is found anyway. Now, if you link to another version (which could very well be, since you seem to be using anaconda) you might run into trouble there.

2000222, 2024-06-10:
Thanks a lot for the response. I checked it and found that this issue does not trouble the subsequent "roslaunch" and "rosrun" robot excuting commands. 😃 So it doesn't matter more.

closed 2024-06-10 by 2000222