UniversalRobots/Universal_Robots_ROS_Driver #706: 'speed_slider_mask' is currently controller by other RTDE client [SW 5.16.1 not supported ?]

niomate, 2024-06-19:
### Affected ROS Driver version(s)

2.1.4

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Windows using WSL2

### How is the UR ROS Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.16.1

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I recently updated the software on the teach pendant of our UR10e to version 5.16.1. 
The connection to the robot worked before that.

## Issue details
Upon starting the driver node/launch file I get the error that " Variable 'speed_slider_mask' is currently controlled by another RTDE client. The input recipe can't be used as configured".
When I searched through the issues, I noticed the Ethernet/IP/Fieldbus issue and checked whether it is somehow related to that, but all Fieldbus options are disabled.
There are other URCaps installed (that should however not impact the speed slider) and we used to have a distance sensor controlling the robot speed, but a) that was removed a few weeks ago and b) it worked with that setup on SW version 5.16.1.

Additionally, the error reads "Please note that the minimum software version required is 3.12.0 for CB3 robots and 5.5.1 for e-Series robots. The error above could be related to a non-supported polyscope version. Please update your robot's software accordingly", which is different from the error mentioned in #204 

## Steps to Reproduce
I don't know whether this is actually reproducable. 
But steps would be to start the driver with polyscope version 5.16.1 (?).

### Relevant log output

```shell
auto-starting new master
process[master]: started with pid [8254]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7f39edcc-2e31-11ef-ac60-00155dfccfd6
process[rosout-1]: started with pid [8275]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [8282]
process[ur_hardware_interface-3]: started with pid [8283]
process[ros_control_controller_spawner-4]: started with pid [8284]
process[ros_control_stopped_spawner-5]: started with pid [8285]
process[controller_stopper-6]: started with pid [8292]
process[ur_hardware_interface/ur_robot_state_helper-7]: started with pid [8299]
process[move_group-8]: started with pid [8302]
[ INFO] [1718797560.520274838]: Waiting for controller manager service to come up on /controller_manager/switch_controller
process[gripper_control_service-9]: started with pid [8307]
[ INFO] [1718797560.521956744]: waitForService: Service [/controller_manager/switch_controller] has not been advertised, waiting...
[ INFO] [1718797560.523514950]: Initializing dashboard client
process[math_service_provider-10]: started with pid [8309]
[ INFO] [1718797560.527342864]: Connected: Universal Robots Dashboard Server

process[unity_debug_output-11]: started with pid [8314]
[... 60 lines cut ...]
[ INFO] [1718797561.394142011]: Looking in common namespaces for param name: flange_manipulator/solve_type
[ INFO] [1718797561.395958417]: Using solve type Distance
[FATAL] [1718797561.934870936]: Variable 'speed_slider_mask' is currently controlled by another RTDE client. The input recipe can't be used as configured
Please note that the minimum software version required is 3.12.0 for CB3 robots and 5.5.1 for e-Series robots. The error above could be related to a non-supported polyscope version. Please update your robot's software accordingly.
[ERROR] [1718797561.934961936]: Could not correctly initialize robot. Exiting
================================================================================REQUIRED process [ur_hardware_interface-3] has died!
process has died [pid 8283, exit code 1, cmd /home/ros/catkin_ws/devel/lib/ur_robot_driver/ur_robot_driver_node __name:=ur_hardware_interface __log:=/home/ros/.ros/log/7f39edcc-2e31-11ef-ac60-00155dfccfd6/ur_hardware_interface-3.log].
log file: /home/ros/.ros/log/7f39edcc-2e31-11ef-ac60-00155dfccfd6/ur_hardware_interface-3*.log
Initiating shutdown!
================================================================================
[unity_debug_output-11] killing on exit
[math_service_provider-10] killing on exit
[gripper_control_service-9] killing on exit
[controller_stopper-6] killing on exit
[ur_hardware_interface/ur_robot_state_helper-7] killing on exit
[ros_control_stopped_spawner-5] killing on exit
[ros_control_controller_spawner-4] killing on exit
[ur_hardware_interface-3] killing on exit
[move_group-8] killing on exit
[robot_state_publisher-2] killing on exit
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-06-20:
Did you check fieldbusses as suggested in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/204?

niomate, 2024-06-20:
If I didn't miss anything, all Fieldbusses (MODBUS, Profinet, Ethernet/IP) are disabled (at least in the teach pendant UI).

urfeex, 2026-05-04:
Sorry that we didn't answer this earlier, but

- The minimum required version output has been updated since then
- The RTDE output conflict occurs only when there's another client claiming it. That could also be another instance of the driver not completely shutdown, another URCap or a fieldbus as mentioned above.

Anyway, I hope this was resolved in the meantime. I'm going to close this, but feel free to comment / reopen if anything is there to add.