UniversalRobots/Universal_Robots_ROS_Driver #710: Unstable connection (UR5 Connection to reverse interface dropped.)

zp2546265641, 2024-07-13:
### Affected ROS Driver version(s)

i do not know which，i download recently from here

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux

### How is the UR ROS Driver installed.

From binary packets

### Which robot platform is the driver connected to.

Real robot

### Robot SW / URSim version(s)

i do not know which

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

i have installed 5.15.76-rt53, but it didn't work as expected. It still reported an error. Whenever I use urcap to connect, start the digital output and analog output IO signals, start my air pump, and then send the trajectory program planned by moveit to the real robot arm, the following problem will occur, but this time after i install LinuxRT kernel it was a little different:
[ INFO] [1720873969.508932034]: Robot connected to reverse interface. Ready to receive control commands.
[ INFO] [1720873975.622664768]: Received request to compute Cartesian path
[ INFO] [1720873975.624387838]: Attempting to follow 31139 waypoints for link 'tool_ros' using a step of 0.010000 m and jump threshold 0.000000 (in global reference frame)
[ INFO] [1720873976.552106491]: Computed Cartesian path with 31145 points (followed 100.000000% of requested trajectory)
[ERROR] [1720873977.292264538]: Sending data through socket failed.
[ INFO] [1720873977.292354495]: Connection to reverse interface dropped.
[ INFO] [1720873977.372595596]: Execution request received
[ERROR] [1720873977.400574305]: Can't accept new action goals. Controller is not running.
[ WARN] [1720873977.400910620]: Controller 'scaled_pos_joint_traj_controller' failed with error INVALID_GOAL:
[ WARN] [1720873977.401011718]: Controller handle scaled_pos_joint_traj_controller reports status FAILED
[ INFO] [1720873977.401071288]: Completed trajectory execution with status FAILED ...
[ INFO] [1720873977.402529277]: Execution completed: FAILED

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-07-17:
This seems to be a duplicate of #709

closed 2024-07-17 by fmauch

zp2546265641, 2024-07-17:
> This seems to be a duplicate of #709

yes,sorry! Because of this, I was troubled for many days and was very anxious.