UniversalRobots/Universal_Robots_ROS_Driver #711: Unable to connect UR5e to a ROS machine due to scaled_pos_traj_controller failing to load

Mechazo11, 2024-07-15:
### Affected ROS Driver version(s)

2.1.5

### Used ROS distribution.

Melodic

### Which combination of platform is the ROS driver running on.

Linux without realtime patch

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

5.11

### How is the ROS driver used.

Others

### Issue details

# Summary
*Short introduction to the issue and how it impact you and why*

Hello,

We have a UR5e mounted on top of a [RBKAIROS](https://robotnik.eu/kairos-mobile-robot/) unit from Robotnik. Due to failure of ```scaled_pos_traj_controller``` to load and initialize, we cannot send commands to UR5e. We can only see its current state in Rviz via MoveIT. 

## Issue details

Network configurations

```
ROS_MASTER: 192.168.0.200
ROS Melodic (cannot be easily upgraded to Noetic)

UR5e IP: 192.168.0.210
Port: 50002
Polyscope 5.11
```

The src file contains many different modules. The UR packages are stored as shown

    └── src/
        └── summit_xl_packages/
            └── manipulation/
                ├── universal_robot
                ├── Universal_Robots_Client_Library
                ├── Universal_Robots_ROS_Driver
                ├── Universal_Robots_ROS_passthrough_controllers
                └── Universal_Robots_ROS_scaled_controllers

Since Robotnik has tightly integrated the UR5e unit with their mobile base, the method of launching the ur driver is slightly different from the methods described in README.md. Here is the error we are getting in the console. The UR ros driver is launched using the following line

```roslaunch summit_xl_bringup arm_complete.launch run_traj_control:=true```

Then I follow the steps to turn on a program with the ExternalControl controller in UR5e`s computer via Polyscope. Then a console in RBKAIROS states ```Robot ready to receive control commands```.

## Steps to Reproduce
Unfortunately, the system is too complex for a simple example

[... 167 lines cut ...]
[ INFO] [1721070811.353852506]: Service available.
[ INFO] [1721070811.353866947]: Waiting for controller list service to come up on /robot/arm/controller_manager/list_controllers
[ INFO] [1721070811.354403369]: Service available.
[INFO] [1721070811.370978]: Controller Spawner: Loaded controllers: pos_traj_controller
[ INFO] [1721070811.474387669]: Robot's safety mode is now NORMAL
[ INFO] [1721070811.477090771]: Robot mode is now POWER_OFF
[INFO] [1721070811.627746]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1721070811.634498]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1721070811.640972]: Loading controller: joint_state_controller
[INFO] [1721070811.659403]: Loading controller: scaled_pos_traj_controller
[ERROR] [1721070811.674409834]: This controller requires a hardware interface of type 'ur_controllers::ScaledPositionJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
[ERROR] [1721070811.674499442]: Initializing controller 'scaled_pos_traj_controller' failed
[ERROR] [1721070812.676216]: Failed to load scaled_pos_traj_controller
[INFO] [1721070812.680056]: Loading controller: speed_scaling_state_controller
[ERROR] [1721070812.690380665]: This controller requires a hardware interface of type 'ur_controllers::SpeedScalingInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
[ERROR] [1721070812.690472386]: Initializing controller 'speed_scaling_state_controller' failed
[ERROR] [1721070813.692026]: Failed to load speed_scaling_state_controller
[INFO] [1721070813.699615]: Loading controller: force_torque_sensor_controller
[INFO] [1721070813.709140]: Controller Spawner: Loaded controllers: joint_state_controller, force_torque_sensor_controller
[INFO] [1721070813.714931]: Started controllers: joint_state_controller, force_torque_sensor_controller
```

### Accept Public visibility

- [X] I agree to make this context public

fmauch, 2024-07-17:
Thank you for reporting this. We are sorry to hear that things don't work for you, as expected.

However, since your UR is tightly integrated into your system, as you wrote, there is not much we can provide in terms of debugging this without any further information.

As the error says:

```
[ERROR] [1721070811.674409834]: This controller requires a hardware interface of type 'ur_controllers::ScaledPositionJointInterface'. Make sure this is registered in the hardware_interface::RobotHW class.
```

Also, your parameters say:

```
/robot/arm/speed_scaling_state_controller/type: ur_controllers/Sp...
```

So, it seems you are loading the controller from the `ur_controllers` package which does not really exist anymore, as things have been migrated to [its own package](https://github.com/UniversalRobots/Universal_Robots_ROS_scaled_controllers/tree/main/speed_scaling_state_controller) (which you seem to have checked out in your workspace, though).

Probably (and this is only guessing since I don't know your setup in detail), you'll need to update your controller.yaml file to make it similar to 

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/497af25ce530ae94a1af04556743e6d3df1eb014/ur_robot_driver/config/ur5e_controllers.yaml#L26-L28

Mechazo11, 2024-07-17:
Hi @fmauch 

Thank you very much for pointing out the EOF of the ```ur_controllers```. I will try to update the ```ur5e_controllers``` file to match the settings shown above.

closed 2024-07-29 by Mechazo11