UniversalRobots/Universal_Robots_ROS_Driver #712: Problem with Connecting UR3 with ROS

MDario1997, 2024-07-16:
### Affected ROS Driver version(s)

Noetic with ubuntu 20.04

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

UR software 3.13.1.10297

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
I am trying to connect the UR3 robot to ROS with the URcap program

## Issue details
I have the UR3 robot on top of a MIR (mobile robot), the UR is connected to the MIR connection, as the computer, because the aim is to connect both to ROS, so i did set all the IP's and after disabling the ufw i launched : 'roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.12.2 kinematics_config:=$(rospack find ur_calibration)/ur3_robot_calibration.yaml' . After doing this, i launched the URcap program on the teach pendant and i have this problem printed on the terminal:
[image]

i don't know if the problem is that the MIR connection is not like ' a classic Wi-Fi' connection, but if i ping both the UR and the MIR on ROS i can see them.

## Expected Behavior
i did expect to just read on the terminal
[ INFO] [1721124400.383903627]: Robot connected to reverse interface. Ready to receive control commands.
waiting then for my commands

### Accept Public visibility

- [X] I agree to make this context public

VinDp, 2024-07-17:
Hi @MDario1997, 
can you give us more details about your network setup? From what you wrote it seems like you are trying to use Wi-Fi instead of cable connection, is that so? Because we do not recommend to use the driver over a wireless connection.

MDario1997, 2024-07-17:
I have an UR3 attached on a moving robot (MIR100), the MIR100 has his own wi-fi, and i connected the UR3 to the MIR (via ethernet) and the host computer to the MIR (via wi-fi, because the robot has to move around the laboratory):
[image]
So with this setup i have the UR3 directly connected to the MIR network via cable, and the computer is connected to the network throu Wi-fi, and if i ping, from ROS, the UR3 IP it works

fmauch, 2024-07-18:
As @VinDp wrote: A WiFi connection isn't sufficient to run the driver. You'll need to have a PC on the robot connected through a cable to the robot in order to run the driver.

MDario1997, 2024-07-18:
Good morning, i tried the setup connecting everything via cable:
[image]
the IP's are: UR3:192.168.12.2, MIR:192.168.12.20,Host PC (selected manually as the tutorial says): 192.168.12.192 .
when i run the command 'roslaunch ur_robot_driver ur3_bringup.launch robot_ip:=192.168.12.2 kinematics_config:=$(rospack find ur_calibration)/ur3_robot_calibration.yaml' , after running the external control program on the teach pensant, i have the same problem as before:
[image]

fmauch, 2024-07-29:
> Edit: are you able to see the picture posted?

yes

What does the log on the teach pendant say? Could you -- for debugging this -- unplug the MIR from the switch to make sure there's not tons of network traffic blocking things from coming through. I don't necessarily expect that, but just to rule out that option.

MDario1997, 2024-07-30:
Good morning,
I solved this initial problem, it was caused by an ethernet port that wasn't functioning properly.
Now if i connect directly the computer to the UR everything works fine, but if i try to create the network between MIR-UR3-PC as shown in the picture before, it works initially, but after some time (5/10 minutes) i lose the connection and i can't send more commands without relaunching the driver:
[image]
i tried to modify i also tried to modify the 'hardwareinterface.ccp' file adding se 'setkeepalivecount(5)' command, but it still an unstable connection. Do you think that is a solvable problem or the fact that another device is connected to the network makes the system always unstable?

tvu-radicalai, 2025-06-16:
Also getting this issue? Has anyone found any fixes to this.

urfeex, 2025-06-17:
@tvu-radicalai please open a new issue clearly describing your setup, the performed commands and their complete output.