UniversalRobots/Universal_Robots_ROS_Driver #717: use moveit to control UR5 's velocity

zp2546265641, 2024-07-23:
### Affected ROS Driver version(s)

master

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux

### How is the UR ROS Driver installed.

Build the driver from source and using the UR Client Library from binary

### Which robot platform is the driver connected to.

UR CB3 robot

### Robot SW / URSim version(s)

3.15.7

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
*I use moveit to control the real robot ur5, but I found that set_max_acceleration_scaling_factor() and set_max_velocity_scaling_factor() do not seem to control the speed of ur5. I set it to 1 but the robot's moving speed did not change. Only the slider on the teach pendant can control the robot's moving speed.

## code
*class MoveItCartesianDemo:
    def __init__(self):

        moveit_commander.roscpp_initialize(sys.argv)

        rospy.init_node('moveit_cartesian_demo', anonymous=True)
        self.arm_group = moveit_commander.MoveGroupCommander('manipulator')

    def move_group_init(self):

        self.arm_group.set_max_acceleration_scaling_factor(1)
        self.arm_group.set_max_velocity_scaling_factor(1)

        self.arm_group.allow_replanning(True)

        self.arm_group.set_pose_reference_frame('base_link') 
        self.arm_group.set_goal_position_tolerance(0.001)
        self.arm_group.set_goal_orientation_tolerance(0.001)*

## Expected Behavior
*i want to set the velocity by the moveit setting, i do not know which cause this problem. If I want to implement this functionality, what do I need to do? 
and what's more, i want to get the tcp's velocity? how can i get it by which function?*

### Accept Public visibility

- [X] I agree to make this context public

zp2546265641, 2024-07-23:
or without moveit? is there any functions else to set the velocity?

zp2546265641, 2024-07-23:
help! my hero!

fmauch, 2024-07-29:
You seem to be posting a lot recently. May I suggest reducing the posting frequency? Our resources to answer to issues are limited which means that it can take a couple of days to respond. Adding additional posts to the issue will not reduce our delay on answering.

What exactly are you trying to achieve? The `set_max_acceleration_scaling_factor` and `set_max_velocity_scaling_factor` functions set the **maximum** speed / acceleration used for generating trajectories from a planned path. That means that the trajectory is generated in such a way, that the robot will not move faster than this throughout the whole trajectory. It does not always run at this speed, as it needs to accelerate / decelerate in order to change its direction.

Please note: This will only change *how fast* a trajectory will be generated. Once the trajectory is generated and sent our for execution, all time, velocity and acceleration constraints of that motions are fixed. The speed slider on the TP can be used to change the speed at any given point in time, but that is something different conceptually.

zp2546265641, 2024-07-30:
> You seem to be posting a lot recently. May I suggest reducing the posting frequency? Our resources to answer to issues are limited which means that it can take a couple of days to respond. Adding [...]

Sorry, I understand. I am still a newcomer and hope to get useful information more quickly through the community. There is no clear answer to this information on the Internet. If someone in the community can give some answers or references, I will be very grateful! What I hope to achieve is that the terminal speed can be obtained and controlled in real time. If it cannot be controlled in real time, I also hope to achieve acceleration in a certain period of time through pre-programmed settings, but I don’t know how to achieve this specifically. Maybe I should learn some other knowledge first?

zp2546265641, 2024-07-30:
> set_max_acceleration_scaling_factor

By the way, there is one more thing. I used the two set statements above, and set a very small number at the beginning, and then directly set it to 1, and ran the action program twice, but the speed of the robot arm did not change significantly, but my other non-UR robot arm did change significantly.

zp2546265641, 2024-08-22:
> You seem to be posting a lot recently. May I suggest reducing the posting frequency? Our resources to answer to issues are limited which means that it can take a couple of days to respond. Adding [...]

Hello, I still have this question. When using urcap to control the operation of a robotic arm, how do I get and set its actual terminal motion speed and acceleration? I know that acceleration and maximum speed can be set using movel on the teach pendant, but I can't find where to set it when using urcap? I even thought about using moveit to set the terminal velocity acceleration, but this can only set the ratio, not the actual effective maximum value.