UniversalRobots/Universal_Robots_ROS_Driver #723: Issue name

komorebiwu, 2024-09-11:
### Affected ROS Driver version(s)

20.04

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux in a virtual machine

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

41.6.4

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

When I start the ur5e mechanical arm to connect with my virtual machine, the following error always appears. I also installed the relevant file package according to the official tutorial, but this problem still occurs. How can I solve it?

### Relevant log output

```shell
[ERROR] [1726062238.586752424]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information] for details.
[ WARN] [1726062238.610965284]: Your system/user seems not to be setup for FIFO scheduling. We recommend using a lowlatency kernel with FIFO scheduling. See https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/real_time.md for details.
```

### Accept Public visibility

- [X] I agree to make this context public

komorebiwu, 2024-09-11:
[image]

urfeex, 2025-05-06:
I know this issue is a bit older, but did you have a look at the link printed in the error output?

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information

That should explain how to extract the robot calibration and use it for an actual robot.