UniversalRobots/Universal_Robots_ROS_Driver #725: Build fail #include <ur_msgs/GetRobotSoftwareVersion.h> not found

LAYERED-pierrechass, 2024-09-20:
### Affected ROS Driver version(s)

ROS Noetic

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Docker

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

UR E-series robot

### Robot SW / URSim version(s)

NA

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
*Short introduction to the issue and how it impact you and why*

## Issue details
When trying to build the driver using the [all source build](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver?tab=readme-ov-file#alternative-all-source-build) the compilation fails with error message: 

`fatal error: ur_msgs/GetRobotSoftwareVersion.h: No such file or directory 57 | #include <ur_msgs/GetRobotSoftwareVersion.h>`
## Steps to Reproduce
Follow instructions in [all source build](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver?tab=readme-ov-file#alternative-all-source-build)

I am wondering if I am doing something wrong here?

Thank you in advance for your help.

### Relevant log output

```shell
<== Failed to process package 'ur_robot_driver': 
  Command '['/workspaces/mas-demo/catkin_ws/devel_isolated/ur_kinematics/env.sh', 'make', '-j22', '-l22']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /workspaces/mas-demo/catkin_ws/build_isolated/ur_robot_driver && /workspaces/mas-demo/catkin_ws/devel_isolated/ur_kinematics/env.sh make -j22 -l22

Command failed, exiting.
bash: devel_isolated/setup.bash: No such file or directory
```

### Accept Public visibility

- [X] I agree to make this context public

sarmadahmad8, 2024-09-26:
having the same issue

fmauch, 2024-09-26:
Is there  a reason why you want to build the driver from source? You can simply install the binary packages using `sudo apt install ros-noetic-ur-robot-driver`

If you really want to build from source:

We recently added a new service to the driver and the message for that is not yet released, hence the build error. You'll need the current version of the `ur_msgs` repo, as well. See #724 

However, we have to update the build instructions for noetic, I think and we might want to use the repos file for that.

The build issue itself will be fixed with the next noetic sync, which should happen somewhere in the next weeks, though.

LAYERED-pierrechass, 2024-09-27:
Hello fmauch,

Thank you for the clarifications and the help! I was trying to build from source because for some reason sudo apt couldn't find the driver, but I then realized I forgot to update apt as I was working on a new docker image, so issue is fixed for me. :)

fmauch, 2024-10-18:
This should be resolved now with the sync happening last night.