UniversalRobots/Universal_Robots_ROS_Driver #730: Issues with Posed Based Cartesian Trajectory Controller

LAYERED-pierrechass, 2024-10-16:
### Affected ROS Driver version(s)

latest

### Used ROS distribution.

Noetic

### Which combination of platform is the ROS driver running on.

Linux

### How is the UR ROS Driver installed.

Build both the ROS driver and UR Client Library from source

### Which robot platform is the driver connected to.

URSim in docker

### Robot SW / URSim version(s)

NA

### How is the ROS driver used.

Through the robot teach pendant using External Control URCap

### Issue details

# Summary
When trying to use the pose_based_cartesian_traj_controller I get an error from URSim stating that the velocity to reach the joint target is exceeding the joint velocity limits followed by an error saying that the robot cannot reach the requested pose. 

## Issue details
I tried to copy paste the example given and change the coordinates to make the point reachable, while it suppresses the 2nd error message about pose not being reachable the first one still remains. I am quite new to all of this and it is hard for me to troubleshoot what's happening. Here is the code I used when trying to input my user defined points:

```
def send_cartesian_trajectory(self):
        """Creates a Cartesian trajectory and sends it using the selected action server"""
        self.switch_controller(self.cartesian_trajectory_controller)

        # make sure the correct controller is loaded and activated
        goal = FollowCartesianTrajectoryGoal()
        trajectory_client = actionlib.SimpleActionClient(
            "{}/follow_cartesian_trajectory".format(self.cartesian_trajectory_controller),
            FollowCartesianTrajectoryAction,
        )

        # Wait for action server to be ready
        timeout = rospy.Duration(5)
        if not trajectory_client.wait_for_server(timeout):
            rospy.logerr("Could not reach controller action server.")
            sys.exit(-1)

        # The following list are arbitrary positions
        # Change to your own needs if desired
        pose_list = [
            geometry_msgs.Pose(
                geometry_msgs.Vector3(0.0, -0.25, 0.8),
                geometry_msgs.Quaternion(-0.7068252, 0, 0, 0.7073883),
            ),
            geometry_msgs.Pose(
                geometry_msgs.Vector3(0.20, -0.25, 0.8),
                geometry_msgs.Quaternion(-0.7068252, 0, 0, 0.7073883),
            ),
        ]
        duration_list = [4.0, 8.0]
        for i, pose in enumerate(pose_list):
            point = CartesianTrajectoryPoint()
            point.pose = pose
            point.time_from_start = rospy.Duration(duration_list[i])
            goal.trajectory.points.append(point)
        goal.goal_time_tolerance = rospy.Duration(100.0)

        rospy.loginfo("Executing trajectory using the {}".format(self.cartesian_trajectory_controller))
        trajectory_client.send_goal(goal)
        trajectory_client.wait_for_result()

        result = trajectory_client.get_result()

        rospy.loginfo("Trajectory execution finished in state {}".format(result.error_code))
```

## Steps to Reproduce

1. Launch URSim in docker with external control enabled as explained [here](https://docs.ros.org/en/iron/p/ur_robot_driver/installation/ursim_docker.html): `docker run --rm -it -p 5900:5900 -p 6080:6080 -v ${HOME}/.ursim/urcaps:/urcaps -v ${HOME}/.ursim/programs:/ursim/programs --name ursim universalrobots/ursim_e-series`
2. Configure URSim (localhost IP, program, ..)
3. Install [Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver)
4. Follow [the instructions to do a test_move](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/doc/usage_example.md#control-the-robot-using-the-test_move-script) and select the pose based cartesian trajectory controller.

## Expected Behavior
That execute the example without any problem.

## Actual Behavior
Error messages (cf pictures)

# Workaround Suggestion
I have tried visualizing the trajectory using the read-only controller as [explained here](https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian/blob/main/cartesian_trajectory_controller/README.md#visualizing-cartesian-trajectories) and it seems correct, however I think there is a discrepancy between the Rviz position/frame of the robot and URSim, I am not sure if it could cause the error or not... 

This issue is more a request for support than anything else as I am sure that I am doing something wrong here but not sure what exactly, it is not clear to me how to use this controller properly.

# Pictures of the error messages:
 [image]
[image]

### Accept Public visibility

- [X] I agree to make this context public

ThomasDuvinage, 2024-10-31:
I confirm to have the same behavior on a real robot for both UR5e and UR10. 

**Affected ROS Driver version(s)**
latest

**Used ROS distribution.**
Noetic

**Polyscope version**
UR5e : 5.14
UR10 : 3.15.4

Which combination of platform is the ROS driver running on.

Linux

## UR10

[image]

## UR5e

[image]

You can check the code to control the robot [here](https://github.com/ThomasDuvinage/ur_robotiq_controller/tree/main/ur_controller).

LAYERED-pierrechass, 2024-11-04:
This actually seems related to this issue? https://github.com/fzi-forschungszentrum-informatik/cartesian_controllers/issues/210

ZhefeiGong, 2024-11-07:
@LAYERED-pierrechass Hey did you find a solution? I had the same problem with the pose based cartesian trajectory controller. Any advice will be appreciated !!!

LAYERED-pierrechass, 2024-11-07:
Hey, unfortunately I only found a workaround by using the JointBasedCartesianTrajectoryController, for my purpose it did the job.